RobotLocomotion/drake #10791: Avoid creating scalar variables when creating new symmetric matrix variables

pangtao22, 2019-02-28:
Consider the following SDP:
```
from pydrake.solvers import mathematicalprogram as mp
import numpy as np

n = 200
Q = np.random.randn(200, 200)
Q = Q + Q.T

prog = mp.MathematicalProgram()
X = prog.NewSymmetricContinuousVariables(n, "X")
prog.AddPositiveSemidefiniteConstraint(X)

for i in range(n):
    prog.AddLinearConstraint(X[i, i] == 1)

prog.AddLinearCost((A*X).sum())

result = mp.Solve(prog)
```

Solving it with Mosek takes a couple of minutes, but it should not take more than a second. The cause is probably that `MathematicalProgram` creates scalar variables in additional to a Mosek "matrix variable", and constrains the scalars to be equal to the entries of the matrix. This is a known issue in JuMP: https://github.com/JuliaOpt/MosekTools.jl/issues/11 . I couldn't print Mosek's log through python, but given the similar amount of slowdown observed in both parsers, I suspect the cause could be similar.

hongkai-dai, 2019-02-28:
@avalenzu this might have a big impact on the computation time for our SOS project. I will work on this before continuing pushing our SOS verification project.

pangtao22, 2019-02-28:
Thanks to @hongkai-dai for binding the terminal output for Mosek. Here's the output for the problem above:
```
Problem
  Name                   :
  Objective sense        : min
  Type                   : CONIC (conic optimization problem)
  Constraints            : 20300
  Cones                  : 0
  Scalar variables       : 20100
  Matrix variables       : 1
  Integer variables      : 0

Optimizer started.
Presolve started.
Linear dependency checker started.
Linear dependency checker terminated.
Eliminator started.
Freed constraints in eliminator : 0
Eliminator terminated.
Eliminator - tries                  : 1                 time                   : 0.00
Lin. dep.  - tries                  : 1                 time                   : 0.00
Lin. dep.  - number                 : 0
Presolve terminated. Time: 0.01
Problem
  Name                   :
  Objective sense        : min
  Type                   : CONIC (conic optimization problem)
  Constraints            : 20300
  Cones                  : 0
  Scalar variables       : 20100
  Matrix variables       : 1
  Integer variables      : 0

Optimizer  - threads                : 8
Optimizer  - solved problem         : the primal
Optimizer  - Constraints            : 20100
Optimizer  - Cones                  : 1
Optimizer  - Scalar variables       : 19901             conic                  : 19901
Optimizer  - Semi-definite variables: 1                 scalarized             : 20100
Factor     - setup time             : 30.13             dense det. time        : 0.00
Factor     - ML order time          : 23.53             GP order time          : 0.00
Factor     - nonzeros before factor : 2.02e+08          after factor           : 2.02e+08
Factor     - dense dim.             : 2                 flops                  : 2.71e+12
ITE PFEAS    DFEAS    GFEAS    PRSTATUS   POBJ              DOBJ              MU       TIME
0   5.0e-01  6.1e+00  1.0e+00  0.00e+00   -3.535989714e+01  -3.535989714e+01  1.0e+00  30.17
1   5.7e-02  7.0e-01  4.1e-02  -9.85e-01  -2.892975415e+03  -2.865197911e+03  1.1e-01  61.30
2   6.7e-03  8.2e-02  6.4e-03  -2.81e-01  -5.963385027e+03  -5.946547072e+03  1.3e-02  91.21
3   7.6e-04  9.3e-03  2.2e-03  7.62e-01   -6.895573531e+03  -6.893787788e+03  1.5e-03  123.08
4   1.2e-04  1.5e-03  9.0e-04  9.55e-01   -7.035287048e+03  -7.035025590e+03  2.4e-04  154.40
5   6.1e-06  7.4e-05  2.1e-04  9.92e-01   -7.061622717e+03  -7.061609762e+03  1.2e-05  185.95
6   2.1e-07  2.5e-06  3.9e-05  1.00e+00   -7.062965917e+03  -7.062965505e+03  4.1e-07  217.91
7   2.9e-08  3.5e-07  1.3e-05  1.00e+00   -7.063007150e+03  -7.063007082e+03  5.7e-08  246.58
8   8.5e-10  1.0e-08  2.4e-06  1.00e+00   -7.063013649e+03  -7.063013647e+03  1.7e-09  277.08
9   2.0e-11  2.0e-09  3.7e-07  1.00e+00   -7.063013847e+03  -7.063013847e+03  4.0e-11  307.39
Optimizer terminated. Time: 307.50
```

pangtao22, 2019-02-28:
But if done right, the output should look more like this: 
```
Problem
  Name                   :                 
  Objective sense        : min             
  Type                   : CONIC (conic optimization problem)
  Constraints            : 200             
  Cones                  : 0               
  Scalar variables       : 0               
  Matrix variables       : 1               
  Integer variables      : 0               
 
Optimizer started.
Presolve started.
Linear dependency checker started.
Linear dependency checker terminated.
Eliminator started.
Freed constraints in eliminator : 0
Eliminator terminated.
Eliminator - tries                  : 1                 time                   : 0.00            
Lin. dep.  - tries                  : 1                 time                   : 0.00            
Lin. dep.  - number                 : 0               
Presolve terminated. Time: 0.00    
Problem
  Name                   :                 
  Objective sense        : min             
  Type                   : CONIC (conic optimization problem)
  Constraints            : 200             
  Cones                  : 0               
  Scalar variables       : 0               
  Matrix variables       : 1               
  Integer variables      : 0               
 
Optimizer  - threads                : 12              
Optimizer  - solved problem         : the primal      
Optimizer  - Constraints            : 200
Optimizer  - Cones                  : 0
Optimizer  - Scalar variables       : 0                 conic                  : 0               
Optimizer  - Semi-definite variables: 1                 scalarized             : 20100           
Factor     - setup time             : 0.01              dense det. time        : 0.00            
Factor     - ML order time          : 0.00              GP order time          : 0.00            
Factor     - nonzeros before factor : 2.01e+04          after factor           : 2.01e+04        
Factor  
```

hongkai-dai, 2019-02-28:
## Problem
The problem is that Mosek solver treats "Matrix variable" in a special manner, while in MathematicalProgram we treat "Matrix variable" as other ordinary variable.

According to https://docs.mosek.com/8.1/capi/tutorial-sdo-shared.html?highlight=msk_appendbarvars, Mosek solves SDP in this primal form
```
min Trace(C, X̅) + cᵀx
s.t lower <= Trace(A, X̅) + aᵀx <= upper
   X̅ is positive semidefinite.
```
where `X̅` is the matrix variable, being positive semidefinite.

On the other hand, MathematicalProgram treats the matrix variable as ordinary scalar/vector variables. So when MathematicalProgram parses the problem to Mosek, in order to conform to Mosek's format, it will introduce N * (N + 1) / 2 scalar variables (call it `z`), with the additional linear constraint `z(i * N + j) - Trace(E_ij, X̅) = 0`, where E_ij is an N x N matrix, whose (i, j)'th entry is 1, and all other entries being 0. (Namely `z(i * N + j) = X̅(i, j)`). So by introducing the new variables `z` and the linear constraint to force z(i * N + j) = X̅(i, j), I can write the problem stored in `MathematicalProgram` in the Mosek format. The price we pay is huge, as we introduce N * (N+1) / 2 new decision variables and linear constraints.

## Solution
I think the right solution is that in `MathematicalProgram`, I should have a new type of constraint for `lower <= Trace(A, X̅) + aᵀx <= upper`. This constraint stores the vector `a`, and the matrix `A` for the constraint. 
 Also I should store a vector of matrix variables `X̅` in `MathematicalProgram`.

@frankpermenter does your new solver also stores everything in the primal form of semidefinite programming? Or do you store it similar to the [SCS format](https://github.com/cvxgrp/scs)

PR #10805 (2019-03-08): Do not add additional variables to Mosek when solving semidefinite program.
Resolves #10791 

cc @pangtao22

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10805)
closed 2019-03-08 by hongkai-dai