RobotLocomotion/drake #10806: doxygen: Explicitly cf. + disambiguate `TypeSafeIndex`, `Identifier`, and `TypeSpecificIdentifier`

EricCousineau-TRI, 2019-03-04:
I knew about `TypeSafeIndex` + `Identifier`, but #10796 showed me that we also have `TypeSpecificIdentifier`.

They all serve distinct purposes, but it'd be nice to ensure that these class have `@see` cross-references with a minor text discussing differences. 

\cc @liangfok @SeanCurtis-TRI

jwnimmer-tri, 2019-03-04:
I would support the automotive one (TypeSpecificIdentifier) having a comment contrasting it with the others.  But I would not want to see the common context (TypeSafeIndex, Identifier) cite the automotive one -- since users of the common ones couldn't / shouldn't typically use the automotive one anyway.

SeanCurtis-TRI, 2019-03-04:
`Identifier` explicitly does that analysis [here](https://github.com/RobotLocomotion/drake/blob/master/geometry/identifier.h#L103) and draws the readers attention to `TypeSafeIndex` [right up front](https://github.com/RobotLocomotion/drake/blob/master/geometry/identifier.h#L18).

`TypeSafeIndex` has the [attention call](https://github.com/RobotLocomotion/drake/blob/master/common/type_safe_index.h#L15) but is missing the section alluded to. That is definitely a bug (and a surprising one to me); I could've sworn it was there.

jwnimmer-tri, 2020-12-09:
The section https://drake.mit.edu/doxygen_cxx/classdrake_1_1_identifier.html#TypeSafeIndexVsIndentifier is working just fine for me.  The TypeSpecificIdentifier has been deleted.  I don't think there's anything remaining to do here.

closed 2020-12-09 by jwnimmer-tri