RobotLocomotion/drake #10816: RigidTransform * Matrix3Xd returns Vector3d

hongkai-dai, 2019-03-05:
There is something incorrect in using operator* with RigidTransform. I tried `X_AB * p_BV`, where `X_AB` is a RigidTransform, and `p_BV` is Eigen::Matrix3Xd with 4 columns. It returns me a Vector3d, only the result of `X_AB * p_BV.col(0)`. I think we should either throw a compilation error or returns a Matrix3Xd. I prefer to returning a Matrix3Xd

sherm1, 2019-03-05:
I marked this high priority -- multiplication is supposed to work right! (attn @mitiguy to investigate please)

EricCousineau-TRI, 2019-03-05:
BTW Does it throw an error in debug mode? I'd assume that it would.

mitiguy, 2019-04-23:
This is being addressed in PR #11308

tri-ltyyu, 2019-04-30:
This issue can close?

mitiguy, 2019-04-30:
This was fixed with PR #11308

closed 2019-04-30 by mitiguy