RobotLocomotion/drake #10818: ipython kernel crashes (with no diagnostics) when input port is not connected

RussTedrake, 2019-03-05:
running this trivial example in a notebook brings down the kernel:
```
from pydrake.examples.pendulum import PendulumPlant

plant = PendulumPlant()
context = plant.CreateDefaultContext()
#context.FixInputPort(0, [0])
derivs = plant.AllocateTimeDerivatives()
plant.CalcTimeDerivatives(context, derivs)
```
uncommenting the FixInputPort line fixes it, of course.

this particular example requires #10817

EricCousineau-TRI, 2019-03-19:
First step will be to see if this happens in C++ too; if so, may pass the buck along :)

EricCousineau-TRI, 2019-03-19:
Can repro in pure C++: https://github.com/EricCousineau-TRI/drake/commit/9c6ea46f52c2048232962357af1b3364db7b17fd
Probably just a port-checking thing.

jwnimmer-tri, 2019-03-19:
Yup.  `this->EvalVectorInput(context, 0)->GetAtIndex(0);` is the old, bad Eval.  If the code used the new port.Eval, this would produce a proper diagnostic.

EricCousineau-TRI, 2019-03-19:
Yup, doing so now.

closed 2019-03-19 by EricCousineau-TRI

akshitj1, 2019-09-11:
Hi @EricCousineau-TRI , still all other examples(acrobot, quadrotor) uses the former  ( `this->EvalVectorInput(context, 0)->GetAtIndex(0);`).  I ran into same issue after making my own plant using those egs. as reference. I was running DirectCollocation on it. Should I send a PR with necessary changes in others too, if trivial?

EricCousineau-TRI, 2019-09-11:
Yup, that would be awesome if you could - thanks!

PR #10955 (2019-03-19): pendulum: Use `InputPort.Eval`
Resolves #10818

Not sure if there's really any value to testing this case.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10955)