RobotLocomotion/drake #10826: Buggy configuration for a Box vs HalfSpace penetration test

amcastro-tri, 2019-03-05:
While debugging #10758 I found a configuration for which the box vs half-space penetration query fails.
You can find a unit test for the `ProximityEngine` [here in my Drake fork](https://github.com/amcastro-tri/drake/blob/19061cebf2fd084db3e38cdf80c8f4ca87e686f7/geometry/test/proximity_engine_test.cc#L2494). You can fetch and compile that for testing with: `bazel run geometry:proximity_engine_test --test_filter=BoxInclinedPlanePenetrationTest*`.

`BoxInclinedPlanePenetrationTest.BuggyCase` is the one test that reports no penetration when there should be penetration. The box for that configuration is deep within the plane. 
The next case `BoxInclinedPlanePenetrationTest.SmallDiffFromBuggyCase` only rotates the box by a very small pitch amount (0.01 degrees) and then we do get the right penetration.

SeanCurtis-TRI, 2019-03-05:
Outstanding! Thanks @amcastro-tri we'll take a look at this. One would presume this is affecting @mitiguy sliding box efforts, yes?

amcastro-tri, 2019-03-05:
Yes, @mitiguy and @sherm1 asked me to debug #10758.

SeanCurtis-TRI, 2019-03-05:
Tangentially related to this: how did you end up noticing the bug, and do you think there's things we can do in the future to help people recognize when these types of bugs crop up?

The first idea that comes to my mind is contact force visualization. Your thoughts?

amcastro-tri, 2019-03-05:
> how did you end up noticing the bug?

The only way I know for this kind of thing that works; dump files with the actual data time step by time step, non-linear iteration by non-linear iteration.

> The first idea that comes to my mind is contact force visualization. Your thoughts?

I was telling @sherm1 and @edrumwri today about this today. For some time I actually thought it was a viz problem, simply because I don't trust the visualizer to render the first few time steps of my sim!! So I do not think the LCM+viz (whatever that viz is) is the proper way to address debbuging tools. All I'd want is to easily introspect ports (ala oscilloscope style with a "probe") and dump that data to a file with a format I could easily specify (ideally columns of numbers). I then can open that with whatever processing tool of my liking. I completely dislike the idea of looking at a plot generated for me (like lcmsocpe) because I do not have control on the very exact point in time (or state or whatever) I want to look into.

PR #10845 (2019-03-07): Correct anchored geometry initialization
The broadphase tree for anchored geometry was treating all anchored
geometry in its canonical frame. This led to erroneous broadphase results
(in which narrowphase results weren't even considered). This corrects
that oversight.

Fixes #10826

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10845)
closed 2019-03-07 by SeanCurtis-TRI