RobotLocomotion/drake #10842: Should we rename StepTo() -> AdvanceTimeTo() ?

sherm1, 2019-03-06:
Because the Simulator advances time in lots of tiny steps of its own choosing, the method name "StepTo()" has created a lot of confusion when we talk about "steps" (most recently this morning with an advanced user). For example, we have "per-step" events and users often think we mean "once per StepTo() call", but that really means "once per trajectory-advancing substep".

What that method actually does is advance time to a given value, using a combination of numerical integration and discrete solvers, using a highly-variable number of steps depending on user choices and runtime behavior of the system.

What do you think about renaming this method `AdvanceTimeTo()` or something like that?

avalenzu, 2019-03-07:
`SimulateTo()`?

edrumwri, 2019-03-07:
That loses some specificity, but does make me think that "AdvanceTimeTo"
might not communicate that state is being advanced.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

amcastro-tri, 2019-03-07:
`AdvanceTo()` (shorter) would do it also IMO.

sherm1, 2019-03-07:
`SimulateTo()` does seem very clear, especially since this is a _Simulator_ method!

amcastro-tri, 2019-03-13:
Reading the `Simulator` documentation I realize we already introduce the terminology [`AdvanceTo`](https://github.com/RobotLocomotion/drake/blob/01490c9dc0756ab4154430b61efc74534f4f454d/systems/analysis/simulator.h#L113) in the documentation's pseudocode. So right now that documentation has a mix of `Advance()`, `Step` and (the ugly) `SubStep`.
Another reason to like `AdvanceTo()` perhaps? that is, easy to explain.

edrumwri, 2019-03-14:
Right. We introduced that term because of the ambiguities induced by
talking about "stepping".

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

sherm1, 2019-03-27:
Status: sherm will add AdvanceTo(), fix the documentation, and convert Drake code to use it. StepTo() will still be available and not officially deprecated since Russ time to adapt.

Per f2f with @edrumwri I will also add `HandlePendingEvents()` as shorthand for `AdvanceTo(context.get_time())` which does the same thing, provided the context is the one in use by the simulator. That is mostly useful in unit testing.

PR #11055 (2019-03-28): Use Simulator::AdvanceTo() instead of StepTo()
Replaces Simulator's StepTo() method (which has caused much confusion by taking an arbitrary number of steps internally) with AdvanceTo() which is more accurate and lets us use "step" in the conventional way meaning just the smallest increment of time advancement.

Also adds an (Advanced) method AdvancePendingEvents() that is equivalent to AdvanceTo(now) and is necessary occasionally, especially in unit tests.

At @RussTedrake's request StepTo() is not yet deprecated but dispreferred.

Resolves #10842

---
This change is [[image]](https://reviewab [... cut]
closed 2019-03-28 by sherm1