RobotLocomotion/drake #10843: Recreate monitoring scripts for queue length, etc. for new Jenkins

jamiesnape, 2019-03-06:


EricCousineau-TRI, 2019-03-07:
Per f2f, this is due to AWS infrastructure flakes that affects Jenkins throughput; AWS had tried spinning stuff up, then Jenkins got into a deadlock.

Workaround is to kill instances that are hanging.

Resolution is to email people, possibly ping `#buildcop` on Slack, when things get backed up.
cref: https://drakedevelopers.slack.com/archives/C270MN28G/p1551459612000400

EricCousineau-TRI, 2019-03-14:
Per f2f: Still tidying up upgrade from old Jenkins => new Jenkins.

tri-ltyyu, 2019-04-11:
This hasn't been breaking CI frequently in the recent weeks. Downgrading priority.

jwnimmer-tri, 2020-12-09:
At this point, things seem stable enough that we don't need to automate the monitoring?  In the rare cases where the queue gets sick, people notice and ping slack, and don't seem to suffer too badly from having to wait the extra time we might have saved by diagnostic the fault earlier in the day.

Please re-open if anyone disagrees.

closed 2020-12-09 by jwnimmer-tri