RobotLocomotion/drake #10850: Technical Debt - proximity_engine.* have become monsters

SeanCurtis-TRI, 2019-03-07:
The `geometry/proximity_engine.*` files have grown out of control. As we support more and more geometry queries (with or without bespoke functionality), they cannot be continually added directly into `proximity_engine.*`. Its growth has led to multiple artifacts:

1. Reduced ability to separate tests and functionality,
2. Increased review cost as we slog through an ever-growing file.

Proposed solution:
 - Create `geometry/proximity/` directory.
 - Move proximity engine into new directory.
 - Each query type's supporting data should go into its *own* .h/.cc files.
 - Unit tests should likewise be decomposed.

The goal is to make it so `ProximityEngine` knows about mapping SceneGraph definitions to various proximity query implementations, but it shouldn't contain all of those implementations.

DamrongGuoy, 2020-02-25:
@SeanCurtis-TRI I think we can close this issue now?

SeanCurtis-TRI, 2020-02-25:
Not yet. While the `proximity_engine.h` is reasonably clean, the unit tests still aren't there. Right now `proximity_engine_tests.cc` still bears a huge burden for testing callbacks. That still needs to move out.

jwnimmer-tri, 2024-03-20:
I am not a fan of "go rewrite some code" tracking issues.  Any time we add new code, we can do some tidy-up of the old code.  If we're not changing the code, then the merit of refactoring by itself is often nost worth the cost.

So, going with "not planned" for now.

closed 2024-03-20 by jwnimmer-tri