RobotLocomotion/drake #10851: Manipulation station needs refactoring to be able to swap grippers readily...

edrumwri, 2019-03-07:
(for enabling a TRI-funded university researcher's project).

EricCousineau-TRI, 2019-03-07:
(Peanut gallery comment) Please make this solution be decomposition + modularity, not more entry points or configurations switches!

sherm1, 2019-03-27:
status: much pain, _almost_ refactored

edrumwri, 2019-03-27:
Correction: it's almost refactored in my branch. Getting it through the PR process may be painful!

EricCousineau-TRI, 2019-03-27:
Is there a WIP branch that the aforementioned peanut gallery can peek at?

edrumwri, 2019-03-27:
I'll post it to this issue when ready. I'm now rejiggering python bindings
for the refactor.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

tri-ltyyu, 2019-04-12:
@edrumwri, any update PR to this issue.

edrumwri, 2019-04-12:
No PR yet, sorting out a final technical wrinkle before I have a running
version locally.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

tri-ltyyu, 2019-04-15:
Is it possible to post the WIP branch to make sure the concerns about modularity raised earlier are addressed?

edrumwri, 2019-04-15:
Yes, but please wait until I've converged on something that runs!

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

edrumwri, 2019-04-17:
Finally got the refactor running today (whew!) I'm going to clean up my code a little and then point interested folks to my branch for comment.

edrumwri, 2019-04-18:
@EricCousineau-TRI @rcory (cc @tri-ltyyu) PTAL at:

https://github.com/edrumwri/drake/tree/manipulation_station_refactor

Unit tests have yet to be modified. Only the end_effector_teleop_mouse is runnable. So this is _not_ ready for PR but will give you the idea of the proposed refactor.

EricCousineau-TRI, 2019-04-22:
I like the manipulator + gripper consolidation, a lot!

My only minor recommendation is that `CombindIiwaWsg` not have any `Iiwa` or `Wsg`-specific methods that are redundant w.r.t. the base interface, unless they're for backwards compatibility and are slated for deprecation.

edrumwri, 2019-04-29:
Status: waiting on @RussTedrake to review the proposed API.

edrumwri, 2019-05-20:
First PR is wending through the review process. See #11320.

sherm1, 2020-02-25:
This looks dead to me. @EricCousineau-TRI @rcory should this be closed now? (Was assigned to Evan.)

rcory, 2020-02-25:
Fine with me to close this out.

sherm1, 2020-02-25:
OK. @EricCousineau-TRI please close this if you think we should or reassign.

EricCousineau-TRI, 2020-02-25:
Will close for now. @edrumwri, please let us know if you might need this in the future!

closed 2020-02-25 by EricCousineau-TRI