RobotLocomotion/drake #10859: multibody: Upstream `ComputeCompositeInertia` from Anzu

EricCousineau-TRI, 2019-03-07:
Follow-up from convos with @amcastro-tri and @RussTedrake 

May relate #10851 (in that you have less book-keeping)

Question is:
Should this be a free function, or yet another method in MultibodyPlant, among 100s of others?

EricCousineau-TRI, 2019-03-07:
Ultimately, the methods I wanna upstream (in some form) are here:
https://gist.github.com/EricCousineau-TRI/d32904bc29cb88d8594631f6e5167cda

sherm1, 2019-03-07:
> Should this be a free function, or yet another method in MultibodyPlant, among 100s of others?

`plant.ComputeCompositeInertia(context, bodies)` is better than a free function IMO. For optimal performance the free function would have to delegate to a plant method anyway, since only the plant has a deep understanding of the exploitable tree topology. It is very common for multibody system classes to have a huge API and I've never seen that cause a problem. Doxygen grouping should help users find the right methods.