RobotLocomotion/drake #10860: Contact results caching causes MBP slowdown?

sherm1, 2019-03-07:
Apparently Drake PR #10783 (cache contact computations) caused a substantial slowdown. Please investigate!

(Observed in Anzu -- see issue [#1952](https://github.awsinternal.tri.global/robotics/enabling-anzu/issues/1952) there.)

/cc @edrumwri

jwnimmer-tri, 2019-03-08:
I also posted this in 1952 but FYI here also -- for now, we're working around this by keeping Anzu's pin of Drake set back prior to #10783, but we can't do that for long before it becomes too painful.

I'd say we either need to merge a fix or work-around today (Friday), or else we revert the Drake PR.  We could even just revert now, if you guys don't mind.  That would give you more breathing room to investigate, instead of a fire drill this morning.

edrumwri, 2019-03-08:
I don't think we should revert to code that was giving wrong answers but
faster.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

jwnimmer-tri, 2019-03-08:
Hmm, ok.  I didn't realize that #10783 was both fixing correctness bugs _and_ trying to improve the performance.

jwnimmer-tri, 2019-03-08:
I see now that the linked issue has more details.  So yes, I agree we should not re-introduce that bug.  I guess it's fire drill time, then.

jwnimmer-tri, 2019-03-08:
Actually, no; on second thought -- the manipulation simulations have been surviving for months with this bug.  After a revert, we can survive for another week while Alejandro et al figures out how to make it both correct and still perform with an acceptable speed.

Anzu is literally unusable from this degree of performance regression.

So I'll return to my original premise -- either the speed gets fixed today, or we revert.

edrumwri, 2019-03-08:
I oppose reverting to buggy code that will impact all of our users, Anzu
and otherwise.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

jwnimmer-tri, 2019-03-08:
I agree with your in principle -- but to reiterate: the performance regression is so bad that the major simulations inside Anzu _are no longer usable_ and the team _cannot do any work_.

Let's see if someone can fix it today, in which case the question is moot anyway.

edrumwri, 2019-03-08:
Immediate fix is necessary, agreed.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

amcastro-tri, 2019-03-08:
Looking!

jwnimmer-tri, 2019-03-09:
@amcastro-tri Can you post a status update?  I have prepared #10875 as a work-around for next week, in case we need it.

amcastro-tri, 2019-03-09:
Two bugs so far:
 1. See #10878.
 2. A small PR that at least bumps Anzu speed up to 1.1X (our target now is 1.6X). I'll PR this soon.

amcastro-tri, 2019-03-09:
Small fix promised in (2) in #10879. This of course does not fix all of the problems we have with discrete updates in #10878 and continuous updates from the integrators as explained in #10879.

jwnimmer-tri, 2019-03-09:
Getting back to 1.1x is plenty good enough to avoid the revert.  Hopefully that can merge before Monday.

sherm1, 2019-03-09:
I think the original Anzu complaint was a 6X slowdown (from 1.8x to 0.3x real time). 1.1x would still be almost a 4X slowdown -- is that really good enough for now?

jwnimmer-tri, 2019-03-09:
Well, of course we'd _like_ to have >1x or even 1.8x, but in terms of a revert, I think our workflows and CI can tolerate sub-realtime.  The bar for keeping users happy is >0.5x worst case.  We can see how stable CI is at 1.1x but my guess is that it's fine.

amcastro-tri, 2019-05-06:
#10889 together with #11125 solve this issue.

closed 2019-05-06 by amcastro-tri