RobotLocomotion/drake #10863: Implement RegionOfAttraction(system, context, options)

RussTedrake, 2019-03-08:
I've been updating all of the sums-of-squares code for underactuated.  While it's fresh in my head, I want to go ahead and knock this long-time missing feature out.

This will be an updated, modernized version of https://github.com/RobotLocomotion/drake/blob/last_sha_with_original_matlab/drake/matlab/systems/%40PolynomialSystem/regionOfAttraction.m

Depends on https://github.com/RobotLocomotion/drake/issues/6732
Depends on https://github.com/RobotLocomotion/drake/issues/10862

RussTedrake, 2019-03-08:
First piece to land will be some visualization helpers.  Something like:

```
PlotEllipse(ax, A,b=[0,0],c=-1, edgecolor, fillcolor) # plots x'Ax + b'x + c = 0 on matplotlib axis.  A must be 2x2
PlotSublevelSet(ax, e, level=1, edgecolor, fillcolor) # plots the sublevel-set of the symbolic::Expression e <= 1.  e must be in two variables.
```
I also want to make sure it's easy to slice/project higher-dimensional elements here.  slice is just Substitute.  Projection needs more care/helpers.

RussTedrake, 2019-03-09:
Variations to (potentially) implement:
- Fixed V, search for ρ 
  - [ ] line search on ρ
  - [ ] Vdot > 0 => V > ρ
- Search for V
  - [ ] bilinear alternations on V, λ
  - [ ] convex outer approximation (e.g. Henrion)

Other options / variations: 
- [ ] Trig-poly w/ s^2 + c^2 = 1
- [ ] Trig poly on the quotient ring
- [ ] Implicit dynamics for rational polynomial (trig/poly) vector fields (e.g. MBP)

RussTedrake, 2019-03-25:
note to self: made progress in `van der pol ROA` notebook, but i need a few of the dependencies to land before i can push much farther.
also waiting on (or need to work around) #11007

RussTedrake, 2019-04-01:
Basic examples working well in python now.

RussTedrake, 2019-07-19:
since this stalled, here is my WIP: https://www.dropbox.com/s/pj0vpl62ywhbbfk/van%20der%20Pol%20ROA.ipynb

sguysc, 2019-08-13:
Is there any modern Drake example of computing funnels for a 
trajectory? (stabilizing a model over a trajectory, tvlqr, finding largest
funnel with time varying Lyapunov functions)

RussTedrake, 2019-08-13:
I’m afraid not yet.  It will be the next on the list once this lands.

RussTedrake, 2020-04-17:
#12827 has landed, and I intend to fill in the different variants.  But I am tracking them in my class todo list, and prefer to track it there instead of here.

closed 2020-04-17 by RussTedrake