RobotLocomotion/drake #10864: Vector output ports with size = 0 have data type set to kAbstractValued

avalenzu, 2019-03-08:
# Problem:
In `CreateCachedLeafOutputPort()` we have the following logic for inferring data type from size:
https://github.com/RobotLocomotion/drake/blob/a7700d3592b72a23cd46e75c9841ae3844bffdb4/systems/framework/leaf_system.h#L2498
This makes it impossible to create 0-element vector output ports, which in turn makes it impossible to connect anything to 0-element vector input ports, for which `get_data_type() == kVectorValued`.

# Proposal:
Pass a `data_type` argument into `CreateCachedLeafOutputPort()` rather than attempting to infer it based on size.

jwnimmer-tri, 2019-03-08:
I have a fix for this in progress.

EricCousineau-TRI, 2019-03-08:
https://github.com/RobotLocomotion/drake/issues/10105 is a duplicate of this.

jwnimmer-tri, 2019-03-08:
Would you marked #10105 as closed now (citing this issue), if indeed its a strict duplicate?

EricCousineau-TRI, 2019-03-08:
Done.

PR #10865 (2019-03-08): framework: Allow for zero-sized ports, state, etc.
Closes #10864.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10865)
closed 2019-03-08 by jwnimmer-tri