RobotLocomotion/drake #10869: Add yaml_read_archive suport

jwnimmer-tri, 2019-03-08:
Anzu has a component named yaml_read_archive, which uses the visitor/archive pattern to easily load yaml data into C++ structures, without having to write boilerplate.

Its API and implmentation seem stable enough now that we should relocate this component to Drake, so it can be used where Drake needs to touch YAML data.

jwnimmer-tri, 2019-03-08:
@sammy-tri My proposed plan would be thus: PR the Anzu files into a `dev` folder in Drake (tests and all) without any changes; this should be a rubber-stamp PR, since the code has already been feature-reviewed.  Then, use Drake PR(s) to finish any TODOs or warts, and then platform-review its move out of dev.  Does that sounds good, or too complicated?

sammy-tri, 2019-03-08:
Sounds reasonable.

PR #10954 (2019-03-21): common/yaml: Mark public and add to Drake install
Closes #10869.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10954)
closed 2019-03-21 by jwnimmer-tri