RobotLocomotion/drake #10878: Simulator logic invalidates ALL discrete events in a diagram.

amcastro-tri, 2019-03-09:
What happens is that when the simulator attempts to handle a single discrete update, all discrete value cache entries in the diagram being simulated get invalidated.

The problematic line of code in the simulator was identified to be within `Simulator<T>::HandleDiscreteUpdate()`, specifically in [this line](https://github.com/RobotLocomotion/drake/blob/109d0564603faf02c95c9e51ac670ab77923414a/systems/analysis/simulator.h#L835), which at the very top level of a `Diagram` invalidates ALL discrete values even if they do not need to be updated.

Relates to #10860. (in that case external publish events at different frequencies is mistakenly invalidating the expensive to compute contact results).

sherm1, 2019-03-09:
Great observation -- this is definitely inadequate as is. The problem is related to #10149 -- we need to be more discerning about what changed before writing everything back. This one's for you, Evan!

edrumwri, 2019-03-10:
Excellent analysis, @amcastro-tri. Is this _the_ performance blocker or only _a_ performance blocker?

amcastro-tri, 2019-03-11:
This is `the` blocker. I'll PR a fix for MBP, but for our most complex sims this starts having a very noticeable impact. I'd push the priority of this issue to the very top of our stack.

amcastro-tri, 2019-03-28:
Sorry, but I got lost in the conversation. Why is it we decided to tackle #10149 first instead of this issue? If I remember correctly this is the cause of slow down in Anzu sims right now.

sherm1, 2019-03-28:
Closing as duplicate of #11059 which has a better discussion thread.

closed 2019-03-28 by sherm1

sherm1, 2019-03-28:
> Why is it we decided to tackle #10149 first instead of this issue?

Seemed easier at the time, but now that we've played with it a bit it doesn't seem to be the panacea we hoped for. Let's attempt the approach outlined in #11059 instead.