RobotLocomotion/drake #10880: Integrators invalidate the entire continuous state + propagate to kinematics values.

amcastro-tri, 2019-03-09:
See description in #10879. I'll paste here for completeness:

> It seems our integrators are broken also. They access (and invalidate) the entire continuous state of a diagram (see here and here for instance).

> Pre this PR, implicit Stribeck cache entries are declared dependent on kinematics_ticket, which essentially gets invalidated if the continuous state gets invalidated. As far as I understand, all systems have a "continuous state tracker" regardless of whether they have continuous state or not. Therefore, when an integrator says "mutate diagram continuous state", that has the side effect of invalidating the kinematics ticket.

> Note that the case we are investigating is that of a discrete state plant (no continuous state at all) however the invalidation triggered from the integrators propagated down the pipeline to something discrete.

sherm1, 2019-03-09:
That's not a problem with the integrators. That's a result of the flawed way we declare state variables (see #9171) which makes it so the kinematics ticket can't tell which states affect kinematics so conservatively is dependent on everything. The workaround is to be more specific in specifying dependencies -- if a cache entry isn't dependent on continuous state variables it should say what it _is_ dependent on rather than using the catch-all kinematics ticket.

It would be a reasonable optimization to avoid propagating continuous state invalidations to subsystems that have no continuous state, in case someone inadvertently declared a dependency on continuous state in that subsystem. All the sources probably should be treated like that (parameters, discrete state, etc.).

edrumwri, 2019-03-09:
Agreeing with @sherm1 here. Given that the description of the issue is exactly what the integrators are supposed to do, I believe this issue should be closed, @amcastro-tri. Do you think agree?

amcastro-tri, 2019-03-09:
I agree, thank you for the second (and third) pair of eyes on this. I therefore bumped the priority for #9171. The "kinematics" state cache work around caused a very difficult to track bug (and maybe there is more?).

closed 2019-03-09 by amcastro-tri

sherm1, 2019-03-09:
I want to capture the idea here of suppressing notifications in some cases as a way to be more relaxed about "all sources" dependencies. I'll file a separate issue.