RobotLocomotion/drake #10882: Suppress diagram out-of-date notifications when we can prove they are no-ops

sherm1, 2019-03-09:
Per discussion in #10880, it is easy for cached computations to end up declaring dependencies on more source values than they actually depend on, since that is the conservative default. That may cause unnecessary invalidations (and hence unnecessary recomputations).

In some cases the Framework could avoid those, without requiring more precise dependency lists. For example, if "all continuous variables" is a dependency, but the subsystem doesn't have any continuous variables, we could just skip that invalidation. Similarly for discrete and abstract state variables and parameters.

Implementation thoughts: Currently a Diagram Context asked to modify continuous state sends a "continuous state changed" notification to every subsystem; it could easily check the continuous state count for a leaf subsystem and skip if zero. An alternative would be for LeafContexts with no continuous states to unregister their continuous state DependencyTrackers so that there is nothing to notify.

jwnimmer-tri, 2021-07-20:
Perhaps with an even broader effect -- the four energy / power trackers are only narrowly useful, but subscribe to _all_ trackers -- including inputs (related to #9171) -- so even a simple PassThrough system is constantly sending invalidation notifications that its potential energy cache is out of date, ditto up through to the Diagram that contains it.  Narrowing the notification tree for stateless systems in particular would be a big win in terms of "pay for only what you need".

jwnimmer-tri, 2024-03-16:
FYI #21134 was related.

sherm1, 2024-03-16:
Between #15432 and #21134 I think we can call this done.

closed 2024-03-16 by sherm1