RobotLocomotion/drake #10886: cpplint: Incorrect parsing in `mathematicalprogram_py.cc`

EricCousineau-TRI, 2019-03-11:
From this Slack post: https://drakedevelopers.slack.com/archives/C3YB3EV5W/p1552112275015500

> @hongkai-dai [1:17 AM]
> In the PR #10873, the CI failed due to the lint test on `mathematicalprogram_py.cc`. But if I run `clang-format-6.0 -style=file -i drake/bindings/pydrake/solvers/mathematicalprogram_py.cc`, the lint test still fails. How should I fix a linting problem for the python binding code?

It looks like `cpplint` struggles with parsing inner arguments to `static_cast<>`, perhaps...

This patch works around it for me:
```
diff --git a/bindings/pydrake/solvers/mathematicalprogram_py.cc b/bindings/pydrake/solvers/mathematicalprogram_py.cc
index b12eec4f0..4bbf12424 100644
--- a/bindings/pydrake/solvers/mathematicalprogram_py.cc
+++ b/bindings/pydrake/solvers/mathematicalprogram_py.cc
@@ -560,6 +560,7 @@ PYBIND11_MODULE(mathematicalprogram, m) {
           py::arg("A"), py::arg("b"), py::arg("vars"),
           doc.MathematicalProgram.AddL2NormCost.doc)
       .def("AddSosConstraint",
+          // NOLINTNEXTLINE(whitespace/parens): Lint bug?
           static_cast<MatrixXDecisionVariable (MathematicalProgram::*)(
               const Polynomial&, const Eigen::Ref<const VectorX<Monomial>>&)>(
               &MathematicalProgram::AddSosConstraint),
@@ -571,6 +572,7 @@ PYBIND11_MODULE(mathematicalprogram, m) {
               &MathematicalProgram::AddSosConstraint),
           doc.MathematicalProgram.AddSosConstraint.doc_1args_p)
       .def("AddSosConstraint",
+          // NOLINTNEXTLINE(whitespace/parens): Lint bug?
           static_cast<MatrixXDecisionVariable (MathematicalProgram::*)(
               const Expression&, const Eigen::Ref<const VectorX<Monomial>>&)>(
               &MathematicalProgram::AddSosConstraint),
```

Don't know the exact error, but filing this issue for traceability / resolution at some point in the future.

\cc @jwnimmer-tri

jwnimmer-tri, 2019-03-11:
This is not the first (nor will be the last) time where a "function pointer type" triggers a false positive in cpplint.  There are already several `// NOLINTNEXTLINE(whitespace/parens)` in `mathematicalprogram_py.cc`.  I would vote for making the work-around more discoverable for @hongkai-dai (if anything), I don't think we should try to fix a regexp-based C++ front-end to avoid corner cases.

jwnimmer-tri, 2019-03-15:
@hongkai-dai Do you have any suggestions for how we could have made the fact more obvious that NOLINTNEXTLINE was a required & expected fix here?  Was there any documentation you looked at (pydrake_pybind.h?) while dealing with this problem?

jwnimmer-tri, 2019-03-15:
Or I wonder... since we have clang-format required in all of pydrake, maybe this rule (or all of the whitespace rules) should be disabled in cpplint for this folder?  What do you think @EricCousineau-TRI?

EricCousineau-TRI, 2019-03-15:
I'd like that solution!

PR #10952 (2019-03-19): pydrake: Disable cpplint warnings subsumed by clang-format
Fixes #10886.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10952)
closed 2019-03-19 by jwnimmer-tri