RobotLocomotion/drake #10888: Create a "discrete contact results" port once #10149 is resolved.

amcastro-tri, 2019-03-11:
Blocked on #10149.

Problem
-----------

Contact results computed by the discrete MBP model should only be valid (and re-calculated) at the periodic time intervals the MBP does discrete updates. This makes sense. Those contact results should only be computed at each discrete update. Anyone looking at those results somewhere within the interval should only see the latest computed contact results, not a new computation for that particular time.

Proposed solution
------------------------
In order to provide this service, we propose the implementation of a **discrete output port** (see [review comment](https://reviewable.io/reviews/robotlocomotion/drake/10879#-L_ZGLphEQfKgIKIFMjf)) for contact results. That is, if anyone looks at the contact results somewhere in between discrete updates they should see the result from the last update. This would be accomplished by having an **unrestricted update** on an abstract state holding the contact results. Essentially this has the effect of a **zero order hold**.

Why didn't we do it yet?
-------------------------------

The reason for not  having this yet in master is #10149. That is, unrestricted updates cause system wide state copies that for most cases of interest (involving geometry) are very expensive and render our simulations useless.

Relates to  #10878, #10880 and #10879

edrumwri, 2019-03-26:
@sherm1 and I just had a very productive discussion about this (in the context of PR #10916). @amcastro-tri you do _not_ want to add abstract state to MBP, if you can help it, and this is one of the cases where you can avoid it. Why? Adding abstract state will keep MBP from being used in the analysis tools (as discussed by @RussTedrake in yesterday's meeting).

The solution to your problem appears to be simple, at least from your standpoint. You need to make the inputs that the implicit Stribeck result is dependent upon emulate a discrete input port by recording the values of those inputs at the discrete update time. Then you will make the implicit Stribeck cache entry be dependent upon those values (as well as the discrete q and v values). I recall you running into an issue with systems external to MBP but internal to its Diagram seemingly spuriously invalidating MBP's discrete q and v values. @sherm1 is looking into that.

sherm1, 2019-03-26:
Say we have a Diagram D with two subsystems, A and B, and both subsystems have some state variables A.x and B.x (may including continuous, discrete, abstract states; i.e., `x={xc,xd,xa}`). Modifying A.xd, for example, will invalidate any computations in A that depend on A.xd. That will _not_ cause invalidation of computations in B that depend on B.xd.

However, modifying D.xd (the composite discrete state at the Diagram level) _will_ generate notifications for A.xd and B.xd which will invalidate computations in both of them. (And note that "modifying" can mean "asking for mutable access".) It is likely this last operation that was causing @amcastro-tri problems, since the Simulator works only at the Diagram level when discrete updates occur.

sherm1, 2019-03-27:
Status: investigated. Looks like a zero-order-hold of the _input_ ports makes more sense and wouldn't require use of the dangerous RawUpdate method.

amcastro-tri, 2019-03-27:
I think the comment [here](https://github.com/RobotLocomotion/drake/blob/master/multibody/plant/multibody_plant.cc#L1774) is wrong. It should say `S⁺ = S(tₛ, x⁻(tₛ), u⁻(tₛ))` (using the notation in `simulator.h`). That is, the Stribeck results `S` should not be looked at as a continuous function of `(t, x, u)` but rather as a discrete update.
This line of reasoning makes me think the solution [here](https://github.com/RobotLocomotion/drake/blob/f3a152c11b8a4f607093bc83009262cca373d392/multibody/plant/multibody_plant.cc#L1791) is already the correct one. 
BTW, zero-order holding the input ports also requires extra state, no?

edrumwri, 2019-03-27:
It does indeed require additional (discrete) state.

jwnimmer-tri, 2019-10-30:
Now that #10149 is resolved, it seems like this issue is unblocked now, and the ticket hack in multibody_plant.cc could be undone.

amcastro-tri, 2019-11-01:
Thank you for the update @jwnimmer-tri. To be honest I'd need to study this again carefully. It looks like a good start is @sherm1's PR #11113?

edrumwri, 2019-12-02:
@amcastro-tri I see that we're both still assigned to this, but I thought that the issue had been resolved through other means. Is this resolving this issue still desired?

amcastro-tri, 2019-12-04:
I'd need help to figure out if #11113 indeed provides what we need. @sherm1, can we discuss tomorrow f2f?

sherm1, 2019-12-04:
Yes, let's discuss. It is now safe to use unrestricted updates if that can be used to resolve this issue.

sherm1, 2020-02-25:
@amcastro-tri is this issue resolved already? Please close if so.

amcastro-tri, 2020-02-26:
My recollection of this problem is summarized in [this note in Drake master](https://github.com/RobotLocomotion/drake/blob/b201621860d8ccbb709ca7fc7dc9534f0b376516/multibody/plant/multibody_plant.cc#L2404). However, per f2f with you @sherm1, you found out that this "hack" relies on caching being enabled, which then led us to believe the correct solution is having a "discrete" input port for everything actuation. That'd require the introduction of additional discrete state variables. Did you create an issue for that conversation that we can refer to? if so, I believe we could close this one for an updated version of this problem, given we probably the solution of #10149 would not solve this problem.

sherm1, 2020-02-26:
@amcastro-tri do you think #12643 covers this adequately?

amcastro-tri, 2020-02-27:
Thanks @sherm1. I attempted to summarize in updated #12786, PTAL.

closed 2020-02-27 by amcastro-tri