RobotLocomotion/drake #10895: Feature request: Warn when LcmSubscriberSystem has a DrakeLcm object with no receive thread.

pvarin, 2019-03-12:
I've run into this bug a couple of times:
1. Setup a system with an LcmSubscriberSystem
2. Forget to call lcm.StartReceiveThread()
3. Wonder why messages contain all zeros

My guess is that it's relatively common and would probably save a lot of debugging time for consumers of the LcmSubscriberSystem we checked that the DrakeLcm object had a valid receive thread when the simulation is run and printed a useful warning message.

jwnimmer-tri, 2019-03-12:
I think the better fix is that LcmSubscriberSystem should not need a thread in the first place.  I will think about whether and how we should do that.

\CC @sammy-tri FYI

jwnimmer-tri, 2019-03-12:
Proposal: add a new class to Drake:
```c++
/// No inputs, no outputs, no state, etc.  Declares _only_ an unrestricted update
/// event that pumps LCM messages into their subscribers iff the LCM stack has
/// message(s) waiting.
class LcmSocketSystem : public LeafSystem {
 public:
   DrakeLcmInterface* get_interface() const;
}
```

Then, simulations look like so:
```c++
DiagramBuilder builder;
// ...
auto* lcm = builder.AddSystem<LcmSocketSystem>();
auto* signal1 = builder.Add(LcmSubcriberSystem::Make<lcmt_drake_signal>("signal1", lcm));
auto* signal2 = builder.Add(LcmSubcriberSystem::Make<lcmt_drake_signal>("signal2", lcm));
builder.Connect(*signal1, ...);
// ...
Simulator simulator(builder.Build());
simulator.StepTo(infinity);
// Look ma, no threads!
```

@pvarin Does this help for your use cases?

jwnimmer-tri, 2019-03-12:
Oops, the new class would be like this, actually:
```
/// No inputs, no outputs, no state, etc.  Declares _only_ an unrestricted update
/// event that pumps LCM messages into their subscribers iff the LCM stack has
/// message(s) waiting.
class LcmSocketSystem : public LeafSystem, DrakeLcmInterface {}
```

jwnimmer-tri, 2019-03-14:
Prototype in progress at https://github.com/jwnimmer-tri/drake/tree/lcm-interface-system.

jwnimmer-tri, 2019-04-18:
I'll guess this is fixed well enough now, due to two major changes:
- As of #11166, `StartReceiveThread` is deprecated.
  - As of #11281, the LcmDrivenLoop will also be deprecated.
- As of #10935, when using LCM in a Diagram the new advice is to do:
```c++
auto lcm = builder->AddSystem<LcmInterfaceSystem>();
auto subscriber = builder->AddSystem(LcmSubscriberSystem::Make<lcmt_foo>(channel_name, lcm));
```
There is no more thread that needs to be started, and by virtue of LcmInterfaceSystem being the only DrakeLcmInterface ever mentioned typically, we're sure to always pass the right one to the subscribers' constructors.

closed 2019-04-18 by jwnimmer-tri