RobotLocomotion/drake #10900: Computing product between two matrices (one containing symbolic::Expression) is very slow

hongkai-dai, 2019-03-12:
@pangtao22 has a problem that solves this problem
```
min trace(Q' * X)
s.t X(i, i) = 1
    X is positive semidefinite
```
where `Q` is a big 200 x 200 matrix. The code is in https://github.com/hongkai-dai/drake/blob/mosek_psd_slow/solvers/test/mosek_psd_demo.cc

Mosek spends just 0.1s to solve this problem, but Drake takes 69s to compute the cost `trace(Q' * X)`. Here is the output of this code
```
computing cost = ((Q' * X).trace()) time (s): 69.311
AddLinearCost(cost) time (s): 0.062083
solver.Solve() time (s): 1.42797
Mosek actual solve time (s): 0.11348
```
The slow code is in this line 
```
  const symbolic::Expression cost = (Q * X.cast<symbolic::Expression>()).trace();
```
I also ran calgrind, here is a screenshot. It seems that most of the time is spent on summing the symbolic expressions.
[image]

cc @soonho-tri what would be a suggested work-around for summing up the expressions? Instead of using symbolic Expressions to compute `trace(Q * X)`, I could write this cost as `a' * x`, and I can explicitly compute `a` without using symbolic Expression, but that is not convenient for the user.

pangtao22, 2019-03-12:
Twan mentioned that JuMP has a special symbolic class for affine functions. It's used to speed things up when polynomial expressions are not necessary. Maybe we could do something similar?

hongkai-dai, 2019-03-12:
BTW, after I replace the symbolic computation
```C++
const symbolic::Expression cost = (Q * X.cast<symbolic::Expression>()).trace();
prog.AddLinearCost(cost);

```
 with the following block
```C++
  Eigen::VectorXd a(n * (n + 1) / 2);
  VectorX<symbolic::Variable> X_lower(n * (n + 1) / 2);
  int lower_index = 0;
  for (int j = 0; j < n; ++j) {
    for (int i = j; i < n; ++i) {
      X_lower(lower_index) = X(i, j);
      a(lower_index) = i == j ? Q(i, i) : Q(i, j) + Q(j, i);
      lower_index++;
    }
  }
  prog.AddLinearCost(a, 0, X_lower);
```
The computation time drops from 69s to 0.008s

jwnimmer-tri, 2019-03-12:
Eigen offers ways to opt-in to "SIMD" approaches to matrix operations, using templates.  By default, it does the naive scalar-wise version; conceivably we could teach eigen how to do smarter "multiple-data" operations on Expression, which could reduce the heap pressure for temporary cells.  Certainly we'd like to be sure that it's only computing the trace elements, not the off-diagonal elements.

pangtao22, 2019-03-12:
I don't think the slowness is entirely the result of computing off-diagonal elements. Profiling result of the following code snippet
```python
from pydrake.solvers import mathematicalprogram as mp
import numpy as np

# Generate Q
n = 200
Q = np.random.randn(200, 200)
Q = Q + Q.T

# formulate and solve SDP relaxation
from pydrake.solvers.mosek import MosekSolver

prog = mp.MathematicalProgram()
X = prog.NewSymmetricContinuousVariables(n, "X")
prog.AddPositiveSemidefiniteConstraint(X)

for i in range(n):
    prog.AddLinearConstraint(X[i, i] == 1)

prog.AddLinearCost((Q*X).sum())

solver = MosekSolver()
solver.set_stream_logging(True, '')
result = solver.Solve(prog)
```
looks like
[image]

soonho-tri, 2019-03-12:
I have investigated this issue. I think this is one of the root causes of the `*` is very slow issues.

Problem
--------

In the current implementation, we do not allow destructive updates on a symbolic expression because of the possible sharing. Let me first illustrate why we wanted the sharing. Consider this example:

```c++
Expression e1 = ...;
Expression e2 = e1;  // Internally `e1` and `e2` share the same cell.
...
// `e1` is updated, but internally we construct a new expression cell and assign it to `e1`
//  to keep `e2` intact.
e1 += ...;  
```

In other words, symbolic `ExpressionCell` is immutable (`const`). Sometimes, this immutability is not desirable. Here is an example illustrating this point:

```c++
Expression e = 0.0;
Expression x = Variable(x);

// e = ∑ i * xⁱ
for (int i = 1; i < N; ++i) {
   e += i * pow(x, i);
}
```

Since `e` is not shared in this example, it should be fine to update the `ExpressionCell` that `e` is pointing to, via in-place fashion. However, we build a new `ExpressionCell` (precisely, `ExpressionAddition`, which includes an `std::unordered_map`, which is not a light-weight object) based on the old `e` and `i * pow(x, i)`.

Solution
--------

I will implement a destructive update for `symbolic::Expression`. The idea is that when we have `e1 += e2` or `e1 *= e2`, first we check if `e1` is shared or not (we can do it via `std::shared_ptr<T>::use_count()`. If it is not shared, we just update the cell in `e1` without creating a new cell.

hongkai-dai, 2019-03-12:
@jwnimmer-tri thanks for pointing out that "trace" issue. Now I replaced the code using trace as `Q * X.cast<symbolic::Expression>()).trace()` with the following block
```C++
  symbolic::Expression cost = 0;
  for (int j = 0; j < n; ++j) {
    for (int i = j; i < n; ++i) {
      cost += (i == j ? Q(i, j) : 2 * Q(i, j)) * X(i, j);
    }
  }
```
which only computes the diagonal terms. This block takes 42s, shorter than the trace version (69s), but still quite slow. 

I think as @soonho-tri pointed out, we should allow destructive update. I see a lot of dynamic memory allocation when I called `(Q * X).trace()`.

jwnimmer-tri, 2019-03-12:
@soonho-tri Instead of (or in addition to?) destructive update, perhaps ExpressionAdd should not always be stored in SoP form?  If it was just an addend chain, you could lazily create (and store) SoP form as soon as some operation required it, but in this trace case it might not even be necessary at all.

soonho-tri, 2019-03-12:
@jwnimmer-tri , that's an option. That will make things significantly faster. One consequence of that approach is that a user needs to explicitly call `Simplify` to cancel / merge terms. For example

```c++
Expression e = 5 * x + y - 3 * x - y;
```

For now, `e` is simplified to `2 * x` immediately. But, under the suggested approach, it will remain as it is, `5 * x + y - 3 * x - y`. I think this is fine, but want to get feedback from other users.

hongkai-dai, 2019-03-12:
Having an unsimplified symbolic::Expression as the input is fine with me. I think in MathematicalProgram, when we parse a linear symbolic Expression as `5 * x + y - 3 * x - y`, we always try to extract the coefficients of each variable, and then store the coefficients in MathematicalProgram constraints. So as long as we can extract the coefficients correctly (which I think we should), I am OK to have an un-simplified symbolic Expression.

soonho-tri, 2019-03-12:
I see. I'll make a version where an addition is represented a pair of Expression (just like an LISP expression, `(+ e1 e2)`). I'll check how far we can go.

jwnimmer-tri, 2019-03-12:
@soonho-tri We could still automatically simplify to SoP on operations like GetVariables, is_polynomial, etc.  In our case (adding a linear cost), it looks like we call GetVariables as we discern its shape, so that is an easy (and precisely correct) time to mutate the internal representation from cons-tree to SoP?

jwnimmer-tri, 2019-03-12:
To be clear, I mean the current Expression semantics and API and unit tests can remain unchanged -- pretend to be simplified / cancellation.  I think we can just lazily defer the canonicalization until we have all of the terms.  MathematicalProgram wouldn't have to change at all.

soonho-tri, 2019-03-12:
@jwnimmer-tri , I think we should keep the API's `symbolic::Expression` intact. But I am not sure if we can keep the unittests unchanged. For example, this test assumes that we do simplification at construction-time: 

https://github.com/RobotLocomotion/drake/blob/111ae16cdf3917f253d5c977715d6d836f0b7dc2/common/test/symbolic_expression_test.cc#L1138-L1139

I also don't think that it's a good idea to mix simplification and other operations (such as `GetVariables`). Especially, I think a getter like `is_polynomial` should be a `const`-operation.

If your point is, "we can embed simplifications in operations like ..., then we can *test* the new implementation without changing MP", I agree with that.

BTW, my biggest concern is that the new representation possibly makes Eigen unhappy which relied on the simplification. That's the first thing I'd check with the new representation.

jwnimmer-tri, 2019-03-12:
My proposal is that factory methods and void methods just cons-tree (do not simplify), but all observers (const getters) **do** internally simplify -- yes, by mutating (or possibly swapping the cell) on a const Expression object.  So because `Expression::EqualTo` simplifies (both terms) first, the test predicate ExprEqual would yield the same answers as it does currently.

jwnimmer-tri, 2019-03-12:
FYI @soonho-tri I think your original "check the update() count" proposal is a lot like https://github.com/simbody/simbody/pull/327 method "detach".  There's every chance that ends up fairly fast.  If it's a more surgical fix that fully deferring the SoP simplification that I propose, and it ends up solving the performance problem, then maybe it's a good place to start from.

soonho-tri, 2019-03-12:
@jwnimmer-tri , OK. I'll implement the destructive-update proposal first. Then, I'll check the example @hongkai-dai found to be slow.

jwnimmer-tri, 2021-07-20:
One FYI with destructive updates -- we should be be careful to consider the effect on `std::hash<Expression>`.  We should be sure that whatever const-ness and/or sole-ownership checks guard the destructive updates are sufficient to prevent set or map keys from changing out from underneath the user.

jwnimmer-tri, 2023-05-04:
With the most recent changes linked above, rather than doing destructive updates, we have overloaded `operator*` to internally accumulate the inner dot products efficiently, and then output the whole matrix multiplication all at once.

We've also bound `pydrake.math.matmul` to void the C++/Python overhead.  Use that function instead of `@` when multiplying large matrices.

Tomorrow (once tonight's nightly builds finish), I'd appreciate if the interested parties upthread could try running the latest nightly build and report back whether or not there are still any performance problems.

If I don't hear anything back, I'll plan to close out this issue in a couple weeks.

jwnimmer-tri, 2023-05-22:
Closing as "good enough" for now.

I am happy to hear about new benchmarks that do not yet run fast enough.  Please open new issues for those.

closed 2023-05-22 by jwnimmer-tri