RobotLocomotion/drake #10905: Segfault when calling ComputeSignedDistancePairwiseClosestPoints

mposa, 2019-03-13:
Trying to running a simple collision detection test between a point (sphere of radius 0) and the ground (modeled as a half space). 

Code is available here:
https://github.com/mposa/dairlib-1/blob/collision-testing/examples/PointPlane/collision_detection_test.cc#L84

Note it's the last line that causes the seg fault.

And the output from valgrind:
```
==3488== Jump to the invalid address stated on the next line
==3488==    at 0x0: ???
==3488==    by 0x4DFE1A: fcl::detail::libccd_extension::__ccdGJK(void const*, void const*, _ccd_t const*, _ccd_simplex_t*) (in /home/posa/.cache/bazel/_bazel_posa/54636c5e9a60e5276059d450b7a41746/execroot/dairlib/bazel-out/k8-opt/bin/examples/PointPlane/collision_detection_test)
==3488==    by 0x4E3921: fcl::detail::libccd_extension::ccdGJKSignedDist(void const*, void const*, _ccd_t const*, _ccd_vec3_t*, _ccd_vec3_t*) (in /home/posa/.cache/bazel/_bazel_posa/54636c5e9a60e5276059d450b7a41746/execroot/dairlib/bazel-out/k8-opt/bin/examples/PointPlane/collision_detection_test)
==3488==    by 0x4E2E05: bool fcl::detail::GJKDistanceImpl<double>(void*, void (*)(void const*, _ccd_vec3_t const*, _ccd_vec3_t*), void*, void (*)(void const*, _ccd_vec3_t const*, _ccd_vec3_t*), unsigned int, double, std::function<double (void const*, void const*, _ccd_t const*, _ccd_vec3_t*, _ccd_vec3_t*)>, double*, Eigen::Matrix<double, 3, 1, 0, 3, 1>*, Eigen::Matrix<double, 3, 1, 0, 3, 1>*) (in /home/posa/.cache/bazel/_bazel_posa/54636c5e9a60e5276059d450b7a41746/execroot/dairlib/bazel-out/k8-opt/bin/examples/PointPlane/collision_detection_test)
==3488==    by 0x4E2F8E: bool fcl::detail::GJKSignedDistance<double>(void*, void (*)(void const*, _ccd_vec3_t const*, _ccd_vec3_t*), void*, void (*)(void const*, _ccd_vec3_t const*, _ccd_vec3_t*), unsigned int, double, double*, Eigen::Matrix<double, 3, 1, 0, 3, 1>*, Eigen::Matrix<double, 3, 1, 0, 3, 1>*) (in /home/posa/.cache/bazel/_bazel_posa/54636c5e9a60e5276059d450b7a41746/execroot/dairlib/bazel-out/k8-opt/bin/examples/PointPlane/collision_detection_test)
==3488==    by 0x3681DA: fcl::detail::ShapeDistanceTraversalNode<fcl::Sphere<double>, fcl::Halfspace<double>, fcl::detail::GJKSolver_libccd<double> >::leafTesting(int, int) const (in /home/posa/.cache/bazel/_bazel_posa/54636c5e9a60e5276059d450b7a41746/execroot/dairlib/bazel-out/k8-opt/bin/examples/PointPlane/collision_detection_test)
==3488==    by 0x4C8A6E: void fcl::detail::distance<double>(fcl::detail::DistanceTraversalNodeBase<double>*, std::__cxx11::list<fcl::detail::BVHFrontNode, std::allocator<fcl::detail::BVHFrontNode> >*, int) (in /home/posa/.cache/bazel/_bazel_posa/54636c5e9a60e5276059d450b7a41746/execroot/dairlib/bazel-out/k8-opt/bin/examples/PointPlane/collision_detection_test)
==3488==    by 0x399A65: fcl::Sphere<double>::S fcl::detail::ShapeShapeDistance<fcl::Sphere<double>, fcl::Halfspace<double>, fcl::detail::GJKSolver_libccd<double> >(fcl::CollisionGeometry<fcl::Sphere<double>::S> const*, Eigen::Transform<fcl::Sphere<double>::S, 3, 1, 0> const&, fcl::CollisionGeometry<fcl::Sphere<double>::S> const*, Eigen::Transform<fcl::Sphere<double>::S, 3, 1, 0> const&, fcl::detail::GJKSolver_libccd<double> const*, fcl::DistanceRequest<fcl::Sphere<double>::S> const&, fcl::DistanceResult<fcl::Sphere<double>::S>&) (in /home/posa/.cache/bazel/_bazel_posa/54636c5e9a60e5276059d450b7a41746/execroot/dairlib/bazel-out/k8-opt/bin/examples/PointPlane/collision_detection_test)
==3488==    by 0x3A8FB2: fcl::detail::GJKSolver_libccd<double>::S fcl::distance<fcl::detail::GJKSolver_libccd<double> >(fcl::CollisionGeometry<fcl::detail::GJKSolver_libccd<double>::S> const*, Eigen::Transform<fcl::detail::GJKSolver_libccd<double>::S, 3, 1, 0> const&, fcl::CollisionGeometry<fcl::detail::GJKSolver_libccd<double>::S> const*, Eigen::Transform<fcl::detail::GJKSolver_libccd<double>::S, 3, 1, 0> const&, fcl::detail::GJKSolver_libccd<double> const*, fcl::DistanceRequest<fcl::detail::GJKSolver_libccd<double>::S> const&, fcl::DistanceResult<fcl::detail::GJKSolver_libccd<double>::S>&) (in /home/posa/.cache/bazel/_bazel_posa/54636c5e9a60e5276059d450b7a41746/execroot/dairlib/bazel-out/k8-opt/bin/examples/PointPlane/collision_detection_test)
==3488==    by 0x3A9C91: double fcl::distance<double>(fcl::CollisionObject<double> const*, fcl::CollisionObject<double> const*, fcl::DistanceRequest<double> const&, fcl::DistanceResult<double>&) (in /home/posa/.cache/bazel/_bazel_posa/54636c5e9a60e5276059d450b7a41746/execroot/dairlib/bazel-out/k8-opt/bin/examples/PointPlane/collision_detection_test)
==3488==    by 0x336296: drake::geometry::internal::(anonymous namespace)::DistanceCallback(fcl::CollisionObject<double>*, fcl::CollisionObject<double>*, void*, double&) (in /home/posa/.cache/bazel/_bazel_posa/54636c5e9a60e5276059d450b7a41746/execroot/dairlib/bazel-out/k8-opt/bin/examples/PointPlane/collision_detection_test)
==3488==    by 0x4C0134: fcl::DynamicAABBTreeCollisionManager<double>::distance(fcl::BroadPhaseCollisionManager<double>*, void*, bool (*)(fcl::CollisionObject<double>*, fcl::CollisionObject<double>*, void*, double&)) const (in /home/posa/.cache/bazel/_bazel_posa/54636c5e9a60e5276059d450b7a41746/execroot/dairlib/bazel-out/k8-opt/bin/examples/PointPlane/collision_detection_test)
==3488==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==3488== 
==3488== 
==3488== Process terminating with default action of signal 11 (SIGSEGV)
==3488==  Bad permissions for mapped region at address 0x0
==3488==    at 0x0: ???
==3488==    by 0x4DFE1A: fcl::detail::libccd_extension::__ccdGJK(void const*, void const*, _ccd_t const*, _ccd_simplex_t*) (in /home/posa/.cache/bazel/_bazel_posa/54636c5e9a60e5276059d450b7a41746/execroot/dairlib/bazel-out/k8-opt/bin/examples/PointPlane/collision_detection_test)
==3488==    by 0x4E3921: fcl::detail::libccd_extension::ccdGJKSignedDist(void const*, void const*, _ccd_t const*, _ccd_vec3_t*, _ccd_vec3_t*) (in /home/posa/.cache/bazel/_bazel_posa/54636c5e9a60e5276059d450b7a41746/execroot/dairlib/bazel-out/k8-opt/bin/examples/PointPlane/collision_detection_test)
==3488==    by 0x4E2E05: bool fcl::detail::GJKDistanceImpl<double>(void*, void (*)(void const*, _ccd_vec3_t const*, _ccd_vec3_t*), void*, void (*)(void const*, _ccd_vec3_t const*, _ccd_vec3_t*), unsigned int, double, std::function<double (void const*, void const*, _ccd_t const*, _ccd_vec3_t*, _ccd_vec3_t*)>, double*, Eigen::Matrix<double, 3, 1, 0, 3, 1>*, Eigen::Matrix<double, 3, 1, 0, 3, 1>*) (in /home/posa/.cache/bazel/_bazel_posa/54636c5e9a60e5276059d450b7a41746/execroot/dairlib/bazel-out/k8-opt/bin/examples/PointPlane/collision_detection_test)
==3488==    by 0x4E2F8E: bool fcl::detail::GJKSignedDistance<double>(void*, void (*)(void const*, _ccd_vec3_t const*, _ccd_vec3_t*), void*, void (*)(void const*, _ccd_vec3_t const*, _ccd_vec3_t*), unsigned int, double, double*, Eigen::Matrix<double, 3, 1, 0, 3, 1>*, Eigen::Matrix<double, 3, 1, 0, 3, 1>*) (in /home/posa/.cache/bazel/_bazel_posa/54636c5e9a60e5276059d450b7a41746/execroot/dairlib/bazel-out/k8-opt/bin/examples/PointPlane/collision_detection_test)
==3488==    by 0x3681DA: fcl::detail::ShapeDistanceTraversalNode<fcl::Sphere<double>, fcl::Halfspace<double>, fcl::detail::GJKSolver_libccd<double> >::leafTesting(int, int) const (in /home/posa/.cache/bazel/_bazel_posa/54636c5e9a60e5276059d450b7a41746/execroot/dairlib/bazel-out/k8-opt/bin/examples/PointPlane/collision_detection_test)
==3488==    by 0x4C8A6E: void fcl::detail::distance<double>(fcl::detail::DistanceTraversalNodeBase<double>*, std::__cxx11::list<fcl::detail::BVHFrontNode, std::allocator<fcl::detail::BVHFrontNode> >*, int) (in /home/posa/.cache/bazel/_bazel_posa/54636c5e9a60e5276059d450b7a41746/execroot/dairlib/bazel-out/k8-opt/bin/examples/PointPlane/collision_detection_test)
==3488==    by 0x399A65: fcl::Sphere<double>::S fcl::detail::ShapeShapeDistance<fcl::Sphere<double>, fcl::Halfspace<double>, fcl::detail::GJKSolver_libccd<double> >(fcl::CollisionGeometry<fcl::Sphere<double>::S> const*, Eigen::Transform<fcl::Sphere<double>::S, 3, 1, 0> const&, fcl::CollisionGeometry<fcl::Sphere<double>::S> const*, Eigen::Transform<fcl::Sphere<double>::S, 3, 1, 0> const&, fcl::detail::GJKSolver_libccd<double> const*, fcl::DistanceRequest<fcl::Sphere<double>::S> const&, fcl::DistanceResult<fcl::Sphere<double>::S>&) (in /home/posa/.cache/bazel/_bazel_posa/54636c5e9a60e5276059d450b7a41746/execroot/dairlib/bazel-out/k8-opt/bin/examples/PointPlane/collision_detection_test)
[... cut ...]
==3488==    by 0x4C0134: fcl::DynamicAABBTreeCollisionManager<double>::distance(fcl::BroadPhaseCollisionManager<double>*, void*, bool (*)(fcl::CollisionObject<double>*, fcl::CollisionObject<double>*, void*, double&)) const (in /home/posa/.cache/bazel/_bazel_posa/54636c5e9a60e5276059d450b7a41746/execroot/dairlib/bazel-out/k8-opt/bin/examples/PointPlane/collision_detection_test)
```

As a secondary quesiton, from this stack trace, it looks like the solver ends up in GJK. Does this mean that point-halfplane is not implemented as a special, and differentiable solver?

Linking in @hongkai-dai  and @sherm1 from Slack conversation.

sherm1, 2019-03-13:
Assigned to @SeanCurtis-TRI, cc @DamrongGuoy for diagnosis.

SeanCurtis-TRI, 2019-03-13:
So, this is the signed distance query.

It's *currently* the case that FCL doesn't have primitive-specific tests for the majority of the primitives for signed distance. We are in the process of rectifying this. As it so happens, you picked on one of the scenarios that falls through to the GJK/EPA implementation (go figure).

That said, a segfault is still quite surprising. We'll take a look.

Big question: without looking at your code (which I will do), are you making queries in multiple threads?

mposa, 2019-03-13:
Thanks. Is there documentation for which FCL queries are supported in a primitive (differentiable a.e.) fashion? This would be very helpful to knowing when we can and cannot use it.

To your question re: threads, no, this was my attempt at putting together the simplest possible query to better understand the API.

SeanCurtis-TRI, 2019-03-13:
No documentation yet. And, ideally, there never would be documentation. After all, the goal is to provide quality answers regardless. However, since this will probably be stalking us for a while, we'll need to do something.

I'm platform reviewer today, but I'll use my free cycles to look into the segfault -- the *worst* that should happen if you use the fall-through path is that you get an answer with bounded error to the true answer. It definitely *shouldn't* segfault.

mposa, 2019-03-13:
It's great for the goal to be quality answers, but sometimes users need to know what's happening under the hood. 

In this example, it'd be nice to know if the answer is an analytical solution or the result of an iterative algorithm.

SeanCurtis-TRI, 2019-03-13:
The challenge of "knowing what's under the hood" is that it can be misleading. There are cases in which the result of an iterative algorithm *is* the correct answer (up to machine precision). So, in writing that documentation, it's important to capture *that* aspect of it rather than the implementation details. In other words, a table that says "for query A, shape B, and shape C, your answer will be the *true* answer to [some precision]" (which would be machine epsilon everywhere we can support it and where we can't, it would be a user-defined tolerance).  So, in that case, the question of *how* the answer is generated is less important than understanding how to interpret it.

But you're right -- documenting that is important. I'm going to post a new issue to capture this need.

SeanCurtis-TRI, 2019-03-13:
See #10907.

SeanCurtis-TRI, 2019-03-13:
FTR I've been able to reproduce your bug and am currently working on flushing it out.

SeanCurtis-TRI, 2019-03-14:
Quick update -- I've found the bug.

The signed distance queries seem to have been an afterthought in FCL. The segfault you experienced is an artifact of that design. I'm going to fix it, but, in the short term, if you replace the half space with an appropriately sized and translated box, you can proceed. After FCL is repaired, we'll update Drake's usage accordingly and then the half space will work. But for now: no signed distance queries with half spaces.

SeanCurtis-TRI, 2019-03-18:
The segfault is no more (replaced with an exception with a meaningful message). However, I'm leaving this issue open as a reminder that we need to support signed distance with halfspaces.

tri-ltyyu, 2019-04-12:
@SeanCurtis-TRI, can i create an issue in FCL (sounds like there's where the halfspace support needs to be) or at least rename this issue and move it back to ready to work so it's clearer what the next step is?

SeanCurtis-TRI, 2019-04-12:
That's a fine idea.

tri-ltyyu, 2019-04-12:
Issue created in FCL https://github.com/flexible-collision-library/fcl/issues/392
This is just a placeholder to link to that. Please feel free to close if it's better to track through FCL only.

SeanCurtis-TRI, 2019-05-13:
Began implementation

SeanCurtis-TRI, 2019-05-14:
We'll consider this closed with the introduction of *any* halfspace signed distance queries. Currently, the signed distance code has limited scope (limited to types of shapes). Getting halfspace from the "never supported" side of the line to the "sometimes supported" is the initial effort. Subsequently, we'll fill in the other (HalfSpace-X) supported shape pairs (where X is more than just Sphere).

PR #11468 (2019-05-15): [proximity] Add support for signed distance to halfspace
1. Extend callback scalar support.
2. Connect pre-existing distance code.
3. Correct assumption about fcl::Halfspace encoding.
4. Add AutoDiff smoke test and scalar support test to distance_to_point_test.
5. Add reality check in proximity engine that a registered halfspace
   produces an answer.

Add support for signed distance between sphere and halfspace

1. Rework scalar support and halfspace detection.
2. Invoke point-halfspace calculation.
3. Extend proximity engine tests

resolves #10905 

---
This change is [[image]](https://revie [... cut]
closed 2019-05-15 by SeanCurtis-TRI