RobotLocomotion/drake #10912: Feature request for contact rich planning

hongkai-dai, 2019-03-13:
As @mposa works on contact implicit trajectory optimization, he mentioned there are some features not available in Drake's MBP+SceneGraph. I put a [doc](https://colab.research.google.com/drive/1J0CxdV7xjHATaimfZLHuwsR8W2E0IOFA) to summarize the needs. It would be great if people could take a look at comment/edit on the doc directly. We will flesh out API design once we collect everybody's opinion on the doc.

cc @mposa @rcory @SeanCurtis-TRI @DamrongGuoy @amcastro-tri @sherm1

sherm1, 2019-03-14:
Cool collab document! I added a small comment.

edrumwri, 2019-03-14:
I added one too, into the text. Is it not possible to link comments to the
individual text?

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

SeanCurtis-TRI, 2019-03-14:
I have to say, I'm not a fan of `colab`. It seems to emphasize `lab` and deemphasize `collaboration`. I think @edrumwri's simultaneous editing with my commenting was interfering with my commenting. And the fact that the comments don't obviously link to text is also inexcusable. You'd expect them to use the google doc APIs. Sheesh...way to go Google.

hongkai-dai, 2019-03-18:
I agree with Sean, that colab is not good for shared editing. I put a summary on the requested feature in this [google doc](https://docs.google.com/document/d/1_WGatuLVUuzaMdApNkZN3Z8J1GSnTf2S5uYUj1_k18s/edit?usp=sharing), can @mposa, @SeanCurtis-TRI and @avalenzu  take a look?

jwnimmer-tri, 2024-03-16:
The call to action here was "look at this google doc and comment".  Looking at the doc, there are comments now, so I'll call this finished now.  It doesn't add any value as a live Drake issue.

closed 2024-03-16 by jwnimmer-tri