RobotLocomotion/drake #10914: Need a non-symbolic way to report if a System is time-invariant.

sherm1, 2019-03-13:
(Requested by @RussTedrake)

Currently SystemSymbolicInspector can be used to determine whether a System is time-invariant, _provided_ (a) the system can be transmogrified to symbolic, and (b) you have time to wait for the answer. For other cases it would be useful to have a `System::IsTimeInvariant()` method analogous to the existing `System::HasDirectFeedthrough()` method, which can be overridden in cases where the System designer can simply provide the answer.

Russ, if you can explain the intended application a little that would be helpful. I'm guessing this is low priority at the moment -- please raise it if it becomes urgent.

RussTedrake, 2019-03-13:
That's a perfect explanation, thanks.

The use case is that we have many algorithms that assume time-invariance.. they should be checking for it.  Here are a few examples I found quickly: 
https://github.com/RobotLocomotion/drake/blob/master/systems/controllers/dynamic_programming.h#L62
https://github.com/RobotLocomotion/drake/blob/master/systems/analysis/lyapunov.h#L27
and there are more on the way.

(I suspect even the trajectory optimization code assumes it).  Probably even `Linearize`.  We haven't had it, and so have been sloppy about this.

sherm1, 2020-02-25:
See also discussion in #12253 and #12709.