RobotLocomotion/drake #10915: Need to be able to force discrete variable update for periodic system

sherm1, 2019-03-13:
(Requested by @RussTedrake)

In PR #6878 @edrumwri provided introspection capability to find out if a discrete System is driven by a single periodic update: [`GetUniquePeriodicDiscreteUpdateAttribute()`](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_system.html#ae0f7d226abe6847e09248db0def0c746).

Next, Russ would like to be able to force that update to occur (see #10909). Calling `CalcDiscreteVariableUpdate()` does not have the intended effect, unless the target System has overridden the discrete variable update event dispatcher, which we tell people not to do. The default dispatcher just runs through the supplied Event objects and invokes their handlers, if present. `CalcDiscreteVariableUpdate()` generates "Forced" Event triggers to supply to the dispatcher. Those normally have no handlers so there is no effect. The target System's event handler that is invoked by the unique periodic update is instead triggered by a "Periodic" Event object, _not_ a "Forced" Event object.

So what's needed is some way to trigger the unique periodic Event (other by running the Simulator past that time), rather than depend on a forced event. Ideas solicited!

I'm not sure about the priority of this. My initial bid is "low" but please adjust.

RussTedrake, 2019-03-13:
Thanks @sherm1.  Low is ok, because I will continue authoring systems using the old event API for now. 

Related: you point out that using `DeclarePeriodicDiscreteUpdate` is no longer the recommended practice. [But a search for it](https://github.com/RobotLocomotion/drake/search?p=2&q=DeclarePeriodicDiscreteUpdate&unscoped_q=DeclarePeriodicDiscreteUpdate) 
turns up many more instances that the comparable search for [`DeclarePeriodicDiscreteUpdateEvent'](https://github.com/RobotLocomotion/drake/search?p=2&q=DeclarePeriodicDiscreteUpdateEvent&unscoped_q=DeclarePeriodicDiscreteUpdateEvent)

we'll also need to look for examples of using [`CalcDiscreteVariableUpdates`](https://github.com/RobotLocomotion/drake/search?q=CalcDiscreteVariableUpdates&unscoped_q=CalcDiscreteVariableUpdates) as many of them may no longer be getting what they expected!

SeanCurtis-TRI, 2019-03-14:
I'll bang my QT-inspired drum for a moment to see if it inspires.

First, QT: it has functions that are, in some sense, declared to be event triggers and event handlers (called signals and slots, respectively). Those are defined as part of the class. But what an trigger kicks off, or what kicks off a handler are *not*.

It would be really nice if, in configuring a diagram, it would be possible to say, "A ha! This has a method for updating discrete state. I'd now like to trigger it for a particular reason and have it execute." It should be a wiring activity and not necessarily an intrinsic property of the system.

I'm now done with my slots-and-signals evangelism. Thank you for your patience.

sherm1, 2019-03-14:
I guess we can do that in this case. If a System has a (signal=periodic trigger, slot=event handler) pair (a.k.a. `Event` object) we need to dig out that pair and dispatch it.

sherm1, 2020-02-25:
See #9171 and [this proposal](https://docs.google.com/document/d/1cTgWeuSqSRg5XyFujzx8ZGdmF9W_3wpp58bQ1xq2zaE) which would fix this issue too.

jwnimmer-tri, 2022-06-28:
We just had a user run into this confusion.

They did the nice thing and authored their example system using `DeclarePeriodicDiscreteUpdateEvent(period, offset, &CalcFunction)` -- using our _recommended_ syntax to declare an update.

But then, when they called `system.CalcDiscreteVariableUpdates(*context, next_state.get());` manually to query what the step would be, it silently did nothing.  This is a big problem!

The work-around for now is to still use the `DeclarePeriodicDiscreteUpdate(period, offset)` to declare the update, even though our API docs recommend against it.

Maybe `DeclarePeriodicDiscreteUpdateEvent` should always add a forced trigger as well?
Or maybe we could have `CalcPeriodicDiscreteVariableUpdates` helper that runs the periodic triggers (only)?

In any case, waiting for #9171 to happen is probably too long of a window.

See also #10909.

sherm1, 2022-06-29:
The two-argument form of `CalcDiscreteVariableUpdates(context, *result)` is badly misnamed. It should be called something obscure like `CalcEffectOfForcedEventsOnDiscreteVariables()`. The actually useful variant is the three-argument form `CalcDiscreteVariableUpdates(context, event_collection, *result)` which could be named something like `CalcEffectOfEventCollectionOnDiscreteVariables()`. The 3-argument form is what the Simulator actually uses -- first it collects the events that have triggered, then using the 3-arg call to calculate their effects, which are then applied to the Context. (The 2-arg form grabs the collection of Forced events and then calls the 3-arg form.) 

An alternate workaround (that doesn't involve overriding the dispatcher) could be to supply the event collection of interest and use the 3-arg form. In this case I believe the author was interested in the effect of a single event.

PR #18106 (2022-11-14): [framework] Deprecate problematic Event API names, replace with better ones
Deprecates problematic Event API names, replace with better ones, modify call sites, improve documentation. Modernizes some examples and tests to use event+handler rather than dispatcher override.

Resolves #10915 

### Background
The Drake Event system and API are an endless source of confusion. In particular, the anachronistic "forced events with no handlers" cause much trouble. Part of the confusion is that methods which trigger only forced events have the same names as general methods that handle modern events (wit [... cut]
closed 2022-11-14 by sherm1