RobotLocomotion/drake #10917: Maliput should support lane directivity.

andrewbest-tri, 2019-03-13:
This issue tracks the implementation of Maliput's DirectionUsageRule.

** Outline **
DirectionUsageRule should contain at-least the following information:
* ID
* LaneSRange
* severity? {Advisory, Strict}
* Direction {WithS, AgainstS, Undirected}

We will add these rules to the RoadRuleBook. This also involves

PR 1
1. Adding the DirectionUsageRule class
1. Adding Comparison methods
1. Adding tests
PR 2
1. Adding getter methods to the RoadRulebook
1. Adding DirectionUsageRule to QueryResults.
1. Updating SimpleRulebook
1. Adding tests
-- 
PR 3
1. Updating Yaml loader to support DirectionUsageRule

** Questions: **
1. Do we need With, Against, Undirected, and Both, or just 3 of the 4
1. Do we need severity? Probably can get by without at first.
1.

andrewbest-tri, 2019-03-13:
@liangfok mind taking a look at this plan and giving feedback?

liangfok, 2019-03-15:
Maliput's design doc includes the following direction usage rules:
 * bidirectional (e.g., non-striped single-lane residential street)
 * unidirectional, s increasing
 * unidirectional, s decreasing
 * bidirectional, turning-only
 * no-traffic (e.g., median strip)
 * parking-lane

What is your reason for simplifying this down to just three?
 * WithS
 * AgainstS
 * Undirected

EricCousineau-TRI, 2019-03-18:
Per discussion in #10930, this design is captured in the maliput org-mode doc here:
https://github.com/RobotLocomotion/drake/blob/c40a01fb7/automotive/maliput/doc/maliput-design.org#L1225

liangfok, 2019-03-26:
@andrewbest-tri can this ticket be closed now that #10951 is merged?

andrewbest-tri, 2019-03-26:
I would like to leave it open until the YAML loader supports DirectionUsageRules.

PR #11228 (2019-04-15): Adds YAML support for DirectionUsageRule.
Resolves #10917.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11228)
commit c16657b1 (2019-04-15): Adds YAML support for DirectionUsageRule.

Resolves #10917.
closed 2019-04-15 by soonho-tri