RobotLocomotion/drake #10918: snopt versions in name?

RussTedrake, 2019-03-14:
Given the variability we are seeing in snopt version performance, is there a way to add more version information into the snopt name returned by MathematicalProgramResults?

Right now `print(result.get_solver_id().name())` just says `SNOPT` always.  But when students/users are getting different results, it would be nice to know which version of snopt they are using.

jwnimmer-tri, 2019-03-14:
I think it would be easy (and uncontroversial) to have `SNOPT/Fortran` and `SNOPT/f2c` names (do you agree @hongkai-dai?).  I am not sure that adding the version number is as easy.  Maybe we could do that in a follow-up, like `class SnoptSolver { public: static std::string version(); };`.

hongkai-dai, 2019-03-14:
> I think it would be easy (and uncontroversial) to have SNOPT/Fortran and SNOPT/f2c names

I think that is good. A follow-up with SnoptSolver::version() is better.

jwnimmer-tri, 2019-03-14:
For version details, we could scrape `README.WhatsNew` to find the string like `Version 7.4-1.1 Elizabeth Wong & Philip E. Gill January 1, 2015.`  It would probably take about 2-4 hours of implementation + review time.  Is that worth it?

hongkai-dai, 2019-03-14:
Personally I won't trust that README.WhatsNew too much. As we know that the SNOPT 7.4 f2c actually uses SNOPT 7.2-12 under the hood.

jwnimmer-tri, 2019-03-14:
I guess we'd be telling the user more about what source archive we used.  They would have to know that `SNOPT/f2c` sources are not described by the README.  Maybe in the f2c case, we should report an empty version string, but in the /fortran case we can report the README?

There isn't a get_version call in SNOPT itself, is there?  I didn't immediately see one.

hongkai-dai, 2019-03-14:
I don't think SNOPT has a get_version call. Actually when I asked Elizabeth how to know the version, she told me to look at the source code to find out the version https://github.com/snopt/snopt-interface/issues/19#issuecomment-409367736

jwnimmer-tri, 2019-03-14:
It looks like with SNOPT/fortran, the print file displays the version number.  Maybe that's good enough to close this issue for now?  WDYT @RussTedrake?

RussTedrake, 2019-03-14:
could we summarize here the steps to having a pydrake user acquire that information (if it's from the print file)?

jwnimmer-tri, 2019-03-14:
They would set an option on the program, like so:
https://github.com/RobotLocomotion/drake/blob/cb651db2aece9c9b24a4ea60e690701b0bc408a6/solvers/test/snopt_solver_test.cc#L141

... or specifically:
```c++
prog.SetSolverOption(SnoptSolver::id(), "Print file", "snopt.txt");
```

Then, after calling Solve(), the file `snopt.txt` has the SNOPT version printed out in the first few lines (and lots of other info, too):

```console
jwnimmer@call-cc:~/jwnimmer-tri/drake$ head /tmp/robotlocomotion_drake_x3pRhv/snopt.out 

         ==============================
         S N O P T  7.4-1.2  (Feb 2015)
         ==============================
1
 
 SNMEMA EXIT 100 -- finished successfully
 SNMEMA INFO 104 -- memory requirements estimated
1
...
```

This only works for SNOPT/fortran.  For SNOPT/f2c, the print file does not appear to contain the "title" (which is what they call the version banner), at least not be default.

RussTedrake, 2019-03-14:
Thx!  Much appreciated!

closed 2019-03-14 by RussTedrake