RobotLocomotion/drake #10932: Compute the gradient of the signed distance / penetration depth normal w.r.t object poses

hongkai-dai, 2019-03-14:
In contact-rich trajectory optimization, we need the gradient of the normal w.r.t the robot configuration. This is because when we impose the friction cone constraint `|fₙ - fₙᵀn*n| ≤ μ fₙ`, (where fₙ is the normal contact force, and `n` is the unit length contact normal), we need to compute the gradient of this function, hence we need the gradient ∂ n / ∂ q. If SceneGraph can report the gradient of the contact normal w.r.t to the geometry poses, then I can compute ∂ n / ∂ q myself.

It seems to me that in general, computing the gradient of the contact normal isn't easy, as the gradient of the normal depends on the curvature of the object at the contact point. So even if we can call FCL using AutoDiffScalar, the automatic differentiation gradient won't contain that curvature information. On the other hand, for the two simple cases (1) sphere-to-sphere (2) sphere-to-halfspace, we can compute the gradient of the contact normal in the closed form. These are the two most commonly used cases. So it would be nice to have a specialized routine for these two cases.

cc @avalenzu @mposa

tri-ltyyu, 2019-03-15:
Required for Q2 (before Q3)

hongkai-dai, 2019-03-21:
Just had more discussion with @SeanCurtis-TRI . I think what we want is differentiable collision query (namely for all the results returned in signed distance query, penetration query, etc, also return the gradient of these results). Currently in the geometry::query_results folder, all of the result (like `SignedDistancePair`, `CollisionPair`, `SignedDistanceToPoint`) are templated, and only instantiated with double. To support gradient info, like `SignedDistancePair<AutoDiffXd>`, we will need to modify the `proximity_engine.cc`. Currently the functions inside `proximity_engine.cc` only supports double, for example https://github.com/RobotLocomotion/drake/blob/06639305ea610f74aac6a4acdd96679af44d4fbf/geometry/proximity_engine.cc#L289-L291

Hence, it seems to me the solution is to change `proximity_engine.cc`, that the functions should also be templated, like 
```C++
template <typename T>
DistanceToPoint(const GeometryId id,
                  const RigidTransform<T>& X_WG,
                  const Vector3<T>& p_WQ)
```

SeanCurtis-TRI, 2019-03-21:
Slight modification:

```c++
template <typename U>
DistanceToPoint(const GeometryId id,
                  const RigidTransform<U>& X_WG,
                  const Vector3<U>& p_WQ)
```

because `ProximityEngine` is, itself, templated on T. And that *might* require some SFINAE to avoid ambiguity...although, I can envision that it might not. That'd be nice. The fact that parameters are typed on the template parameter suggests you should be good.

sherm1, 2019-03-21:
Having the method itself templated would mean that all of its implementation would have to be in headers. That seems extreme. Maybe just have an API that returns the gradients?

hongkai-dai, 2019-03-21:
Had another f2f discussion with @sherm1 @SeanCurtis-TRI and @DamrongGuoy , we think the right approach is not to templatize proximity_engine. Instead we will define a new query result type, that contains the gradient w.r.t the relative pose between the two geometries.
```C++
struct SignedDistancePairWithGradient {
  GeometryId id_A;
  GeometryId id_B;
  Eigen::Vector3d  p_ACa;
  Eigen::Matrix3d dp_ACa_dp_AB; // The gradient of the witness point p_ACa w.r.t the object B's position in A's frame.
  Eigen::Matrix<double, 3, 4> dp_ACa_dquat_AB; // The gradient of the witness point p_ACa w.r.t object B's orientation in A's frame.
  Eigen::Vector3d p_BCb;
  Eigen::Matrix3d dp_BCb_dp_AB;
  Eigen::Matrix<double, 3, 4> dp_BCb_dquat_AB;
  double distance;
  Eigen::RowVector3d ddistance_dp_AB;
  Eigen::RowVector4d ddistance_dquat_AB;
  Eigen::Vector3d nhat_BA_W;
  Eigen::Matrix3d dnhat_BA_W_dp_AB;
  Eigen::Matrix3d dnhat_BA_W_dquat_AB;
};
```
There are two arguments not to templatize proximity_engine

1. It is too much work to templatize this file, with more than 1700 lines.
2. Computation with AutoDiffXd is not very efficient. In this case we know we always take the gradient w.r.t the relative pose between the two object size, hence we can do all the computation with fixed size gradient.

avalenzu, 2019-03-25:
Will it be an issue that we don't have a method to compute `dquat_AB_dq` in MbP?

hongkai-dai, 2019-03-25:
> Will it be an issue that we don't have a method to compute `dquat_AB_dq` in MbP?

Good call. I think we have dangular_velocity_AB_dq in MBP, and we can compute dqdot_dangular_velocity, so we can compute dquat_AB_dq, but it would be better if MBP provides a function to do that.

avalenzu, 2019-03-25:
>  we can compute dqdot_dangular_velocity, 

Do you mean dquatdot_dangular_velocity?

hongkai-dai, 2019-03-25:
> Do you mean dquatdot_dangular_velocity?

Yes, good catch.

hongkai-dai, 2019-05-20:
This is resolved.

PR #11086 (2019-04-02): Add signed distance from a point to sphere and halfspaces, together with its gradient
This is the first step to resolve #10932.

I will add the query to return `SignedDistancePairWithGradient` in the next PR.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11086)
closed 2019-05-20 by hongkai-dai