RobotLocomotion/drake #10934: Poor collision checking performance

calderpg-tri, 2019-03-15:
Right now, collision checks using SceneGraph use one of two options:

- `ComputePointPairPenetration()` for binary checks

- `ComputeSignedDistancePairwiseClosestPoints` when gradients or padding are required

Using `ComputeSignedDistancePairwiseClosestPoints` is significantly more expensive, since it performs N^2 checks, but is the only way to implement collision avoidance in controllers or variable padding around the robot/environment.

In practice, many uses of `ComputeSignedDistancePairwiseClosestPoints` look like this:
```
const auto& query_object = GetQueryObject();
std::vector<SignedDistancePair<double>> signed_distance_pairs = query_object.ComputeSignedDistancePairwiseClosestPoints();
for (const auto& signed_distance_pair : signed_distance_pairs) {
  const double distance = signed_distance_pair.distance;
    if (distance <= influence_distance) {
      // do work
    }
}
```
where only those pairs within an influence distance are used.

We should offer a version of `ComputeSignedDistancePairwiseClosestPoints()` that takes an influence distance parameter such as `ComputeSignedDistancePairwiseClosestPointsWithinDistance(double distance)` so that we can take advantage of broadphase to reduce the expense of collision checking.

calderpg-tri, 2019-03-15:
Marked as high priority since this affects RobotBridge collision avoidance and the performance of all planners in anzu.

sherm1, 2019-03-15:
This makes a lot of sense and seems like it would be easy to add? @DamrongGuoy built that feature 
into the third available query:
```c++
  std::vector<SignedDistanceToPoint<double>>
  ComputeSignedDistanceToPoint(const Vector3<double>& p_WQ, double threshold);
```

Assigned to @DamrongGuoy and @SeanCurtis-TRI for investigation/discussion.

tri-ltyyu, 2019-03-15:
Important for dish demo. Have benchmark use cases for testing. Should be low amount of work.

jamiesnape, 2019-03-18:
(As fundamental as collision checking is, this is surely high priority for a toolkit such as Drake regardless of Anzu.)

SeanCurtis-TRI, 2019-03-18:
@jamiesnape FTR the issue name is *slightly* misleading. This isn't strictly about performance of the collision checking system. It's about a paucity of options in the API. In other word, @calderpg-tri has been forced to swat flies with a sledge hammer. Adding the parameter won't affect users that do collision detection, or users that require O(N^2) signed distance results. So, it's not a general "performance" tweak that helps the world at large. However, it is a valuable API with good utility.

jamiesnape, 2019-03-18:
> the issue name is slightly misleading

Would be good to fix that. Does not look so great on the first page of issues for someone looking at the toolkit for the first time...

sherm1, 2019-03-18:
Good idea @jamiesnape, done.

SeanCurtis-TRI, 2019-05-15:
Resolved by #11429.

@calderpg-tri can you confirm and close this issue?

SeanCurtis-TRI, 2019-05-20:
Exercising this new functionality seems to have [revealed a problem](https://github.awsinternal.tri.global/robotics/enabling-anzu/pull/2318). Won't be closing this PR until that is resolved.

jamiesnape, 2019-05-20:
Can you summarize the problem instead of posting an intranet link?

SeanCurtis-TRI, 2019-05-20:
@jamiesnape sorry about that.

Essentially, for a given configuration, using a threshold of zero, two passes on the what should be the same data produce different answers. I'm not entirely clear on the full characterization of the bug. But that's the crux.

PR #11508 (2019-05-21): bugfix - Signed distance queries with a zero threshold lose results
FCL is fundamentally incompatible with a zero-valued threshold. The
broadphase has a *special* function for traversing for distance. This
function assumes that it is only reporting *separating* distance. As such,
it only computes box *separating* distance and doesn't even take action if
the boxes are intersecting.

So, in order to avoid falling into that beartrap, we make sure that we
always pass in a positive tolerance and rely on Drake-side culling to
eliminate any distance results that don't satisfy the threshold.

resolves [... cut]
closed 2019-05-21 by SeanCurtis-TRI