RobotLocomotion/drake #10965: move mathematical program jupyter notebook to drake

RussTedrake, 2019-03-20:
From https://github.com/RussTedrake/underactuated/blob/master/src/mathematical_program_examples.ipynb
I could imagine it living in `drake/tutorials` or `drake/doc/tutorials`.

Depends on #9650

EricCousineau-TRI, 2019-03-20:
My vote is that it go under either `drake/doc/tutorials`, or possibly even `drake/examples`?

We can always figure out the install story for notebooks 0then (which is still a bit murky pending resolution of stuff like #9645).
Absolutely, we should *never* vendor in `jupyter`. Any platform that doesn't have the deps to support it natively doesn't get to use it (which I think is only Xenial, and will disappear from support soon-ish anywho, per #10606.)

RussTedrake, 2019-03-20:
`drake/examples` is also fine for me.  I slightly prefer that over `drake/doc/tutorials`... I want these to be easily findable.
what exactly do you mean by "vendor" in jupyter?

EricCousineau-TRI, 2019-03-20:
By "vendor", I mean that we install our own version of `jupyter`.

(I was / am still entertaining the idea of vendoring in `numpy` for #8116; even though it's a core package it seems to be excellent in terms of ABI backwards compatibility + minimal disruption, and has minimal disruptions. I have not seen this in Jupyter; I also haven't had to dig in either.)

EricCousineau-TRI, 2019-09-19:
Done via #11694

closed 2019-09-19 by EricCousineau-TRI