RobotLocomotion/drake #10971: AutoDiffXd implementation messes up with caching in odd ways.

amcastro-tri, 2019-03-20:
Uff, this is a tough one, I'll try my best to explain. The buggy code is in #10959 (the two CI failures).
The failure can be reproduced when running one of the inverse kinematics tests with:
```
bazel run -c dbg multibody/inverse_kinematics:minimum_distance_constraint_test
```

### Observation ###
A cached value that should never change after allocation (first entry in [this pool](https://github.com/RobotLocomotion/drake/blob/b1593027de27ab7ac3a25e2b7acb2f330f6b3c7c/multibody/tree/position_kinematics_cache.h#L123), `world_index()` is always zero.) does happen to change. In particular, what changes is the size of the derivatives vector when using AutoDiffXd. The cache entry is constant, with zero (or none) derivatives. However somewhere, somehow, someone manages to bypass our beautiful `get_mutable_foo()` pipeline.

### How to track ###
Run under your debugger, I used gdb and I'll post the steps here.

1. `gdb bazel-bin/multibody/inverse_kinematics/minimum_distance_constraint_test`
2. run
3. where.
4. Identify the frame calling `RigidTransform::operator*()`. Long story short, `p_AoBo_A_`  comes from that cache entry I mentioned at the top. It should never change since its allocation†. (for me this is frame 16)
5. Get the address of the derivatives of `p_AoBo_A_`. Issue: `p &p_AoBo_A_[0].derivatives()`.
6. Place a watchpoint (defaults to track mem writes). Issue: `watch *0x55555730a0f0` (or whatever address you got in step 5).
7. Rerun. Issue: `run`.
8. The watchpoint will trigger about 4 times before getting to that particular point given that that piece of memory happens to be used for other stuff at the beginning. Then it gets triggered the first time the cache entry is allocated (**good**) and then yes finally when our memory changes again (**bad**, it should be const!).
9. **Identify who changes it.**. If you follow to the top of the call stack, it boils down to [this line in `common/autodiffxd.h`](https://github.com/RobotLocomotion/drake/blob/b1593027de27ab7ac3a25e2b7acb2f330f6b3c7c/common/autodiffxd.h#L188).

So the `internal::make_coherent(m_derivatives, other.derivatives());` is the problem. If you follow its impl in `Eigen/src/AutoDiff/AutoDiffScalar.h`, you'll see a **const cast** is issued in order to make derivative sizes "coherent".

† Long story. You can go down the stack call and identify the line of code in [`body_node.h:1365`](https://github.com/RobotLocomotion/drake/blob/b1593027de27ab7ac3a25e2b7acb2f330f6b3c7c/multibody/tree/body_node.h#L1365) responsible for updating the position kinematics. In that line you'll see `X_WB = X_WP * X_PB;`, where `X_WP` is a **const** reference. When calling `RigidTransform::operator*()` however, some magic happens and the derivatives of `X_WP` change (well, their size, they still are zero).

### So again, how is this a problem? ###
well, in step 9 above we still didn't trigger the original error, we just saw how something supposed to be const changes.
What happens next is that the unit test `minimum_distance_constraint_test` runs two cases with the same `Context`. One with derivatives of size 14 first followed by a case with derivatives of size 2. When we start, from a zero sized derivatives in `X_WP` to a "coherent size" in `internal::make_coherent`, everything is good.
The problem is that next time Eigen tries to use `X_WP` for computations involving derivatives of size 2, it "believes" `X_WP` is non-constant since it now has derivatives of size 14 (not 2!). So when trying to perform the math in `RigidTransform::operator*()` Eigen now finds `p_AoBo_A_` with derivatives of size 14 (from previous test, even if still zero) and `p_BoQ_B` with derivatives of size 2 (from the new run, ok). **Now Eigen barks the two things have different size!!**

### Why didn't we see this before? ###
My guess is that `Eigen::internal::make_coherent()`'s templatized implementation was run with different types before (while Isometry internally is a 4x4 matrix, RigidTransform is more like a Vector3 plus a Matrix3x3).

jwnimmer-tri, 2019-03-20:
I am well-versed in `make_coherent`'s evil `const_cast` (see #6944).

I see two options:

(1) When the user changes the size of their derivatives vector, they need to CreateDefaultContext and start over.  Once tainted by any derivatives.size(), you can't change sizes for that context.

(2) Change drake/common/autodiffxd.h to not call `make_coherent` anymore, but instead add runtime checks for each term whether `term.derivatives.empty()`, and if so use `VectorXd::Zero(other_num_derivatives)` instead of writing the zeros-vector after a const_cast.  This might even make us much faster because we can short-circuit the chain rule etc. and never touch the heap when the values in a particular expression have empty derivatives.

SeanCurtis-TRI, 2019-03-20:
I was thinking a variation of the first option (I was thinking more of a "flush derivatives" method, but both lead to a "start afresh" state). I like the second option and would never have come up with it because I'm not so well versed with drake's autodiff fun.

jwnimmer-tri, 2019-03-20:
I think (1) is a fair work-around for that unit test, if you have a MBP PR that's blocked on this @amcastro-tri.

@SeanCurtis-TRI Nice, we'll label "context.FlushDerivatives" as (3), then.  It's a fine choice in case (2) turns out to be more difficult then I imagine.

I'll also offer (4) -- entirely replace AutoDiffXd with our own implementation.  I have a half-baked one already in an old branch, where I was playing with doing the small-buffer optimization on it.  (When you only have 10-ish derivatives (6? 14?) you shouldn't need to hit the heap, they can live on the stack.  This is useful for small, fast control problems, and/or when using chunking.)

amcastro-tri, 2019-03-20:
Thank you guys! yez, I think option (2) would be the quickest and offers some sort of correctness (modulo AutoDiffXd internal hacks). 

I love option (4) @jwnimmer-tri! I was wondering how much of a big project would be getting that SSO in our `autodiffxd.h`, given we already have a Drake copy anyways.

amcastro-tri, 2019-03-20:
Ok, I just verified this solves the problem:
```
template <typename OtherDerType>
  inline const AutoDiffScalar<DerType> operator+(
      const AutoDiffScalar<OtherDerType>& other) const {
    if (m_derivatives.size() == 0) {
      return AutoDiffScalar<DerType>(m_value + other.value(),
                                     other.derivatives());
    } else if (other.derivatives().size() == 0) {
      return AutoDiffScalar<DerType>(m_value + other.value(),
                                     other.derivatives());
    }

    internal::make_coherent(m_derivatives, other.derivatives());
    return AutoDiffScalar<DerType>(m_value + other.value(),
                                   m_derivatives + other.derivatives());
  }
```

how do you guys like it?

SeanCurtis-TRI, 2019-03-20:
My mental model must be flawed. I don't see how this solves the problem. I see that it *does* make the implicit promotion cheaper (by not actually promoting for this particularly amenable operation). But if the two vectors still have different-sized derivatives, your'e still calling `make_coherent` and trying to add vectors of different sizes. How does that not still blow up in the case documented above?

amcastro-tri, 2019-03-20:
As @jwnimmer-tri pointed in Slack, the `make_coherent` is a no-op for two non-empty vectors. That is, we don't even need it with this fix. 
If you get into a situation where those two scalar types have different sized derivatives, then your bug is somewhere else in your setup. I updated the snippet by removing the `make_coherent` call.

amcastro-tri, 2019-03-20:
when I run `bazel run -c dbg common:autodiffxd_test` with the proposed fix, some of the tests do not pass. I do realize however that they correspond to cases with non-existent derivatives.
For instance, the case `log(x - x) + (y - y)`, but the derivative of `log(x-x)` doesn't exist. Are those tests cases intentional?

jwnimmer-tri, 2019-03-20:
How do they fail?  My first impression is that they should be valid.

One effect of this change might be that values used to have `.derivatives().size() > 0` but they were all zeros, but now they are empty.  Is the test case just overly constrictive?  Might be easier to use slack for debugging.

amcastro-tri, 2019-03-20:
I pushed #10972 only for discussion. But as you can see, the tests in master were broken. Our friendly `make_coherent` was coupling all those tests since they "remembered" previous results.

PR #10975 (2019-03-26): Stop promoting zero sized derivatives.
That is, we stop using `Eigen::internal::make_coherent`.
This resolves #10971 and fixes the unit tests as proposed in #10972.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10975)
closed 2019-03-26 by amcastro-tri