RobotLocomotion/drake #10973: [dynamics] MBP document relationship between multibody frame and scene graph frames

SeanCurtis-TRI, 2019-03-20:
## Problem statement

Given an MBP and SG instance populated with frames and geometries, downstream users can query an individual geometry's fixed pose relative to its SG frame (i.e., `X_FG`). There is *some* relationship between `F` (the geometry's parent SG frame) and `B` the body frame in MBP. However, this relationship isn't documented anywhere.

## Proposal

My current understanding is that the registered geometry frame is coincident and aligned with the corresponding body frame. This should be captured in documentation. The logical place to do this will be in `MBP::RegisterGeometry()` (as this is where the frames are registered).

cc @calderpg-tri @siyuanfeng-tri @amcastro-tri

amcastro-tri, 2019-03-20:
There is [this](https://github.com/RobotLocomotion/drake/blob/9a902a6be1a6a0edafb305f92615d16cd523bed3/multibody/plant/multibody_plant.h#L2364) but I guess you want to say more? in particular, how do you retrieve `X_FG` from scene graph these days? maybe we can link that API in MBP?

SeanCurtis-TRI, 2019-03-21:
I basically want to document something very specific: for every SG frame that is registered, it is coincident and aligned to the corresponding body frame. So, `X_BᵢFᵢ = I`.

PR #11873 (2019-08-01): Elaborate documentation relating body frame with scene graph frame
This makes it clear that when MBP registers a frame in SceneGraph for a body, that the two frames are coincident and aligned.

resolves #10973

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11873)
closed 2019-08-01 by SeanCurtis-TRI