RobotLocomotion/drake #10976: SceneGraph: Remove `X_PG` (feature to spec. geometries w.r.t. other geometries)

EricCousineau-TRI, 2019-03-20:
Per f2f with @calderpg-tri:
It's a bit confusing to see `X_PG` in `SceneGraph`-related, when the docs and code sometimes criss-cross between `X_PG` and `X_FG`.

To me, it seems like a generally redundant quantity, and I think we should remove it. User should attach geometries to frames only, always. If you want some more relative pose stuff, then you add another frame.

(I say this b/c I've been working towards removing redundant frames from Anzu code + models... I also like deleting redundant code!)

Per f2f with @SeanCurtis-TRI, he seems game for this too.
(Assigning it to you for now, but I'm up for reassignment.)

\cc @scpeters @azeey (only on account of this relating to recent discussions on SDF geometries + frames)

SeanCurtis-TRI, 2019-03-21:
FTR I only partially agree with what you've written. The part I *don't* agree with is:

> If you want some more relative pose stuff, then you add another frame.

All frames registered to SG are assumed to move. So, going this route for what should be a fixed frame artificially and meaninglessly inflates the data that has to be reported to SG and the computation that SG does in preparation for the various black boxes that service queries.

That said, I don't know that I buy the problem statement and how it maps to the solution:

1. I've striven to document `X_FG` and `X_PG`. If there are issues with documentation, that should be addressed. Ultimately, I don't see that the two transforms should interfere with people's work (nothing prevents the workflow you've outlined). And I believe that any confusion can be alleviated in documentation. Certainly statements like "...and code sometimes criss-cross between..." is a defect in the code and should be called out specifically. 
2. Arbitrarily excising code based on principle is a bit like introducing code based on principle. 

I would push this *very* far down the priority list -- well after goemetry/dev is removed.

EricCousineau-TRI, 2019-03-21:
> FTR I only partially agree with what you've written. The part I don't agree with is:

Dern! Almost conscripted you to my side :P

> I would push this very far down the priority list -- well after goemetry/dev is removed.

Absolutely agreed! Hence `priority: low` + `cleanup`.

> Certainly statements like "...and code sometimes criss-cross between..." is a defect in the code and should be called out specifically.

Shall do! Will see if I can find those bits, maybe ping @calderpg-tri for providing what he's seen as well.

> Arbitrarily excising code based on principle is a bit like introducing code based on principle.

IMO: Disagreed! Excising code admits simpler APIs; at present, the APIs still require a level of expert knowledge (e.g. Calder and I had to read through b/c we wanted to learn more about collision filtering), which involves reading public and internal APIs. Redundant features slow down my ability to understand what's going on.

Every line of code and documentation is a potential point of failure. Having fewer unnecessary lines makes things potentially simpler.

That being said, yes, I should fully understand what's there is a necessary exercise (and what workflow, if any, this *really* supports); fixing minor / acute flaws is *potentially* a valuable exercise.

SeanCurtis-TRI, 2019-03-21:
I humbly retract the

> Arbitrarily excising code based on principle is a bit like introducing code based on principle.

That was definitely a bridge too far. It is definitely unused functionality and, as such, runs the risk of becoming zombie code. It does increase complexity and I'm sorry for it serving as a speedbump in the road to understanding. However, there's a cost to remove it. And if it turns out it *is* a useful API, the cost to put it back is not a simple revert, it'll have to be redone because SG itself will have moved on.

Generally, we seem to be on the same page -- the *value* of this API is unproven. It leads to subtlety and nuance that slows down readers. Unless a meaningful use case crops up between now and the time it seems prudent to remove it, we remove it.

SeanCurtis-TRI, 2019-05-17:
## New proposal

Reduce confusion via code and documentation *placement*. 

Essentially, it comes down to this. There is value in registering articulated frames as children of other frames and geometries on other geometries. However, there is no general necessity. So, the proposal is to re-arrange and re-document the SceneGraph public API so that the typical user would think it about it as a world frame with a flat list of child frames (the moving, register frames), each with a list of zero or more rigidly-affixed geometries. This is where the concepts of `X_WF` and `X_FG` are relevant. 

In a different part of the code, the ability to register a geometry to a geometry or a frame to a frame are put in an "Advanced" section which introduce the concepts of the more elaborate hierarchy and the concomittant poses `X_PF` and `X_PG`. 

The fundamental functionality will not change, but they'll no longer be as strongly co-located reducing immediate cognitive dissonance. This also means I won't have to spend time stripping out code that has been tested and exercised for a year and a half.

Thoughts?

calderpg-tri, 2019-05-17:
While I think the latest proposal would improve the situation, I am still dubious about the value of registering geometries to other geometries. @EricCousineau-TRI Is this functionality we ever want to support in SDFormat?

If the answer is no, I see no reason to support "advanced" behavior with limited use cases that cannot be parsed and instead must be created via code.

If the answer is yes, then we should change the API so `X_FG` always does what you want - fixed pose relative to a frame, either that of the geometry itself or that of its parent geometry, so that `X_WG = X_WF * X_FG` and it is easy to check if the geometry is registered to a frame or another geometry (perhaps by having `X_PG` return an optional or pointer).

sherm1, 2019-05-17:
Sdf is not the only way people can use Drake, so I don't that is the right yardstick. A hierarchy of frames is basically the _definition_ of a SceneGraph, so we would need to make a good case that Drake's SG doesn't need that functionality (and perhaps rename it to avoid false advertising). Since it would now be a lot of work to rip out this functionality from SG, that would need to get scheduled as a project competing with other much higher-priority projects. 

It would be much easier just to hide that in an Advanced section of Doxygen, and make sure we have a nice API for supporting the currently-dominant use of flat frames, where a user would not have to deal with any of the fancy functionality.

From my reading of the OP, I believe that would address the presenting problem?

calderpg-tri, 2019-05-18:
> A hierarchy of frames is basically the definition of a SceneGraph, so we would need to make a good case that Drake's SG doesn't need that functionality

That is not in question. What is in question is if *geometries* can be registered to *both* frames and other geometries, or just frames. We aren't trying to build the best SceneGraph generally, we're trying to build a SceneGraph that meets the needs of drake, and that means deciding if the ability to have
```
frame->geometry->geometry
```
versus
```
frame->geometry
   |->geometry
```
is worth the confusion of separate `X_FG` and `X_PG`.

If it is worth it, then we need to spend the time to make the API make sense and do the right thing and fix all the uses in drake and anzu that expect `X_FG = X_PG`. If it's valuable enough to be worth our time to make work properly, then it's valuable enough to support through the parser.

On the other hand, if this is an obscure "feature" we hide from normal users because we don't think people will actually use it, then we pay a maintenance cost and a correctness cost for supporting something that never gets used, and we should remove it.

SeanCurtis-TRI, 2019-05-20:
I think the fundamental issue is one of over-simplification and aliasing.

Ultimately, `SceneGraph` deals with two types of frames: articulated and fixed. Articulated frames must have poses provided for them on `SceneGraph`'s input ports. Fixed frames are hard-coded and immutable (and geometric shapes are strictly associated with fixed frames). The distinction at declaration is important because it defines what it means for a port value to be valid and the entity registering geometry needs to have a clear understanding of what it is committing itself to.

In principle, I can build a hierarchy of these frames. The world frame is the canonical fixed frame and I can hang either articulated frames or fixed frames on the world frame. Articulated frames can be hung on other articulated frames. And I can hang fixed frames on articulated or fixed frames. (Currently, the only fixed frame that an articulated frame can be hung on is the world frame -- this prevents it from being a completely open set up -- one can argue this is an undesirable inconsistency.)

This is where the aliasing comes in. The "articulated frames" are called `GeometryFrame` with corresponding `FrameId`s. The "fixed frames" are called `GeometryInstance` with corresponding `GeometryId`s (with an implicit pose mapping the geometry's *canonical* frame C to the fixed frame G: `X_GC = I`).

Perhaps it would've been better to simply allow "articulated" and "fixed" frames to be declared explicitly and then to *always* define the pose `X_GC` when registering geometry such that the pose `X_GC` is a constant value but the frame `G` can be either fixed or articulated. Then frames' parents are *always* frames. Geometry parents are always frames. And there is an arbitrary chain from any frame to the world and, similarly, from a geometry to the world (this would also implicitly allow a path from an articulated frame to the world that includes fixed frames).

rpoyner-tri, 2025-09-09:
bouncing to new component owner @SeanCurtis-TRI for re-triage.