RobotLocomotion/drake #10982: py mbp: Changing bindings to use `RIgidTransfrom`

EricCousineau-TRI, 2019-03-21:
Per this conversation:
https://reviewable.io/reviews/robotlocomotion/drake/10959#-LaQLvhdDZ03DikBQUnY

May depend on viability of turning of implicit casts.

\cc @amcastro-tri

EricCousineau-TRI, 2019-03-27:
Hm... I updated the checkboxes for what would maximize backwards compatibility. However, given that we've already broken it (as some Anzu code is broken by the current Drake `master`), should we just keep it as-is? I can forward-fix the Anzu code.

@jwnimmer-tri @amcastro-tri Can I ask y'all's opinion?

jwnimmer-tri, 2019-03-27:
If doing the same thing in pydrake as we did in C++ is easy enough (and it looks like that's what your checkbox list implies), then that seems worthwhile.

jwnimmer-tri, 2022-05-03:
@EricCousineau-TRI I think this work might already be finished?  If not, could you clarify what remains?

EricCousineau-TRI, 2022-05-03:
Ah, nope - it seems to be all done!

closed 2022-05-03 by EricCousineau-TRI