RobotLocomotion/drake #10984: [geometry] QueryObject::ComputeSignedDistanceToPoint() bad for multiple points

SeanCurtis-TRI, 2019-03-21:
## Problem Statement

Due to legacy reasons that have yet to be corrected, every time *any* `QueryObject` method is invoked, it does a full pose update on *all* of SceneGraph. That means if multiple points need to be queried, the results will be needlessly slowed down.

## Proposal

Actually making the update properly conditioned on the cache entries which track whether the work has already been done is a ways down the road. The simplest solution is to provide a parallel implementation that takes a *vector* of points and returns a *vector* of results, such that it does the full pose update once for the entire set.

Users of this method should weigh in on how important this is.

/cc @calderpg-tri

DamrongGuoy, 2019-03-21:
Thank you @SeanCurtis-TRI .  That's very insightful.  I agree taking in a set of query points instead of one single query point is an improvement.  We will have to discuss the API.

calderpg-tri, 2019-03-21:
Can this be linked with #10442 so that in the case of a "frozen" context the pose update is not performed? The usage I have in mind would be:

1. Update `q`

2. Perform dummy `ComputeSignedDistanceToPoint` query to update SceneGraph

3. Freeze context

4. Do a bunch of queries

5. Unfreeze context

A batch API, while definitely nice to have, doesn't fit the needs of our collision checkers well.

SeanCurtis-TRI, 2019-03-21:
I'd propose a simple extension of the existing API.

Current API:
```c++
std::vector<SignedDistanceToPoint<double>>
  ComputeSignedDistanceToPoint(const Vector3<double> &p_WQ,
                               const double threshold
                               = std::numeric_limits<double>::infinity()) const;
```

Multi-query API:
```c++
std::vector<std::vector<SignedDistanceToPoint<double>>>
  ComputeSignedDistanceToPoint(const std::vector<Vector3<double>> &p_WQs,
                               const double threshold
                               = std::numeric_limits<double>::infinity()) const;
```

As to @calderpg-tri's question, this is related but somewhat orthogonal to freezing. This is a manifestation of `SceneGraph` cheating and going around the system. Even with context freezing, `SceneGraph` will not work naturally with the frozen context (doing some cache entry hacks, would work). There's still some more investigation that needs to go into what the right short-term solution for making `SceneGraph` sufficiently well-behaved.

It's good to know that batch isn't going to cut it for your application. So, in the absence of a user clamoring for this feature, it should remain low-priority in favor of making sure it plays nicely with frozen contexts.

calderpg-tri, 2019-03-21:
I'm all for a batch API, and it's probably easy enough to implement that we should do it now even if no one uses it.

My thinking on frozen contexts is that in lieu of proper caching, we just assume/enforce that a frozen context "must be true" and never do any updates if the frozen bit is set. It's hacky and unpleasant, but probably gives us the short-term performance boost.

sherm1, 2019-03-21:
Caching is so dangerous that I don't think anything hacky should be involved. SG should use the cache, and the cache can have a safe "frozen" feature that will barf if an update is needed. Only the cache has the infrastructure needed to do that flawlessly.

calderpg-tri, 2019-03-21:
I agree that this *should* be done right. But we don't want the perfect to be the enemy of the "good enough for a specific carefully-documented use-case" when this is (the?) most expensive operation during collision checking and collision checking speed is a bottleneck for pretty much every planning algorithm ever written.

SeanCurtis-TRI, 2019-03-21:
In consultation we've come up with a solution that should satisfy all parties.  But first, some background. 

`SceneGraph` is weird because it has a bunch of state-ful black boxes and predated the cache system by a fair margin. One artifact is what is listed above in the main description of this issue. Another important artifact is that `SceneGraph` will *not* play nicely with a frozen context. At least, in its current state.

The solution will be a coarse approximation of what `SceneGraph` will look like down the road when we make it cache-compliant. Essentially, we create a cache entry for "black boxes updated". `Calc`'ing that cache entry has the side effect of pulling poses and updating all the black boxes (in the future we'll have finer granularity). This will have two effects:

1. It will play nicely with the frozen context -- even to the extent that if you forget to evaluate a query *prior* to freezing, you'll get feedback.
2. It will also improve performance w.r.t. the current single-point query. The first query on dirty inputs will update the black box, but subsequent queries will not.

(Captured in #10988.)

calderpg-tri, 2019-03-21:
That solution sounds good to me.

SeanCurtis-TRI, 2020-09-25:
There hasn't been much clamoring for the batch API. The cost of calling multiple queries (needlessly duplicating the work of updating poses) was eliminated as of #11019. So, I'm closing this for now. If we want the batch API, we can resurrect this (or create a new issue).

closed 2020-09-25 by SeanCurtis-TRI