RobotLocomotion/drake #10990: [geometry] Provide domain of signed distance to point queries

SeanCurtis-TRI, 2019-03-21:
## Problem Statement

The various queries in `QueryObject` currently span the world. All geometries/geometry pairs which have not been explicitly excluded via collision filters are considered. These methods were intended to be the first-pass, generic interface from which all other, more limited, queries can be derived. However, there is a very real computational cost in stripping out unneeded results. This is the first of several additions to the API to allow more targeted queries.

In this first pass, we want to specify some *subset* of geometries to query signed distance to a point.

## Proposal

1. Mark `ComputeSignedDistanceToPoint()` as deprecated.
2. Add  `ComputeSignedDistanceWorldToPoint()` as the replacement of the old method.
3. Add the new method `ComputeSignedDistanceSetToPoint()` with the following interface:

```c++
/** Query in the signed-distance-to-point query family. Distances are only
   computed to the geometries contained in the given set rather than all
   geometries in the world (given supported shape types).  If `geometries`
   is empty, no work will be done and the returned vector will likewise
   be empty.  */
  std::vector<SignedDistanceToPoint<double>> ComputeSignedDistanceToPoint(
      const Vector3<double>& p_WQ, const GeometrySet& geometries,
      const double threshold = std::numeric_limits<double>::infinity()) const;
```

calderpg-tri, 2019-03-22:
If possible, we'd like an equivalent function
```
std::vector<SignedDistanceToPoint<double>> ComputeSignedDistanceToPointExcluding(
      const Vector3<double>& p_WQ, const GeometrySet& geometries_to_exclude,
      const double threshold = std::numeric_limits<double>::infinity()) const;
```
since it can be easier to know which geometries you *don't* want than the geometries you do want (especially if you're adding/removing new geometries).

SeanCurtis-TRI, 2019-03-23:
I don't see that that should be a problem -- and the "exclusion" variant is the higher priority?

calderpg-tri, 2019-03-25:
I think both inclusive and exclusive variants have equal priority.

An alternative would be to provide a wider range of set-like operations on top of `GeometrySet` so that inclusion/exclusion could be expressed there instead.

SeanCurtis-TRI, 2019-03-25:
Can you elaborate on your last point?

calderpg-tri, 2019-03-25:
My thinking there is that `GeometrySet` could behave more like a proper set (it already isn't intended to be iterable), and allow expressions like this
```
SceneGraph().ComputeSignedDistanceToPoint(p_WQ, NotIn(geometries), threshold)
```
where the exclusion could be accomplished by the `NotIn` rather than a separate function on SceneGraph. The natural extension of this would be other set operators:
```
SceneGraph().ComputeSignedDistanceToPoint(p_WQ, Union(geometries_1, geometries_b), threshold)
```
```
SceneGraph().ComputeSignedDistanceToPoint(p_WQ, Intersection(geometries_1, geometries_b), threshold)
```
Something like this could be accomplished in a variety of ways (inheritance, std::functions, etc).