RobotLocomotion/drake #10992: Integrator statistics need to be adjusted when caching is enabled

edrumwri, 2019-03-21:
Integrators use a special derivative evaluation function to compute statistics over ODE calls. This function needs to be modified so that it understands when the derivative evaluation is cached and the ODE is not newly re-evaluated.

sherm1, 2019-03-26:
FYI @edrumwri I determined experimentally that at least for the existing rk3 and implicit_euler test cases the Simulator never tries to evaluate the derivative twice with identical contexts.

edrumwri, 2019-03-26:
Simulator never tries or the caching system never tries?

sherm1, 2019-03-26:
Every call to `IntegratorBase::EvalTimeDerivatives()` done by the integrators caused an actual computation indicating either that (a) something in the context had changed since the last call, or (b) caching was not enabled. Looking at the tests it appeared that caching was manually enabled for almost all of them but I might have missed something. When #10979 lands (caching by default) we can retry but by inspection I couldn't actually spot a case where the integrators do repeated derivative calculations.

edrumwri, 2019-03-26:
I'm missing something then because we're relying upon the caching system to keep from repeating the first-same-as-last computation: how could "every call ... [cause] an actual computation" and the "integrators [not] do repeated derivative calculations"?

sherm1, 2019-03-26:
Which integrators actually compute derivatives after computing the final state value?

edrumwri, 2019-03-26:
It depends how many steps you're taking. But regardless, there appears to still be a logical conflict in your statement above: either calls cause repeated computation or they don't (unless you're referring to a specific case, like existing tests, and not the general behavior).

edrumwri, 2019-03-26:
Update per f2f: RK3 is not actually a FSAL integrator (contrary to its documentation).

PR #11028 (2019-03-27): Avoid double-counting for cached derivative evals in integrators
Resolves #10992
- Adds (Advanced) method to get time derivative cache entry.
- Use that method to avoid double counting of cached evaluations in integrators for statistics reporting.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11028)
closed 2019-03-27 by sherm1