RobotLocomotion/drake #10996: .pyc files in source directories

RussTedrake, 2019-03-22:
Running `bazel run //doc:serve_sphinx` results in `doc/sphinx_base.pyc` being left in the source dir (for git to complain about).

Running tests in manipulation station now results in `examples/manipulation_station/differential_ik.pyc` being left in the source dir.

Although I'm tempted to add the `*.pyc` to `.gitignore`, there is probably a better solution?

jwnimmer-tri, 2019-03-22:
I started seeing this too, and was annoyed.  I only now just realized that my `realpath` call in #10969 is likely at fault.

jwnimmer-tri, 2019-03-22:
Actually, maybe it was tickled by my test runner change, but it was a bug in #10632 as it turns out.

jwnimmer-tri, 2019-03-22:
The sphinx pyc file thing has been a bug for a while, I think it was @EricCousineau-TRI 8502b6167910e9cad74fd059f4fe04288d27b654 which did it.  I'll let him decide how to mitigate / ignore it.

EricCousineau-TRI, 2019-03-22:
In the FWIW category, I always add `*.pyc` files to my global `gitignore` file.

But yeah, we shouldn't be spewing these out in the first place!

EricCousineau-TRI, 2019-03-22:
Confirmed that this also happens with `--config=python3`; in this case, it pukes `__pycache__` all over the workspace.

Seems to be related to https://bugs.python.org/issue6386 (though probably not due to this bug, as the author wants to do weird things with relative imports?)

Made a repro as well, here: https://github.com/EricCousineau-TRI/repro/commit/c9135157f589d757ab9861cb53301f1932f70521

```
$ ./symlink_junk.sh python2
...
+ python2 ./symlink/root.py
...
+ find . -name '*.pyc' -o -name __pycache__
./sub.pyc

$ ./symlink_junk.sh python3
...
+ python3 ./symlink/root.py
...
+ find . -name '*.pyc' -o -name __pycache__
./__pycache__
./__pycache__/sub.cpython-36.pyc
```

This is mentioned tangentially here: https://github.com/bazelbuild/bazel/issues/793#issuecomment-172259206
Since `bazel run` runs without sandboxing, it spews junk everywhere.

I don't feel like doing any Bazel gymnastics to suppress this right now, as it's due to really esoteric behavior on Python's side for doing weird crap with symlinks. (I don't like this feature, and also really don't like that Python permits importing relative modules at all...)

(My ideal idea of behavior is that either (a) executing the symlink just straight up fails. Failling that, I'd prefer (b) that the compiled bytecode litter the symlink tree instead...)

Will just submit a PR to suppress in `.gitignore`; maybe letter we can try suppressing bytecode, but meh for now?

EricCousineau-TRI, 2019-03-22:
Er, looking at #10997, will first try absolute importing (using the behavior that I hate... implicit `__init__.py` files...)

PR #10998 (2019-03-23): doc: Use absolute imports
Resolves #10996

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10998)
closed 2019-03-23 by EricCousineau-TRI