RobotLocomotion/drake #11005: Installation issue about drake

ddliugit, 2019-03-23:
While I followed the installation procedure of drake on the website, I met an error. 

This is the link I followed:
http://underactuated.csail.mit.edu/underactuated.html?chapter=drake

This is the error I met when I went to the step A.2.3 Test for success:

10-16-79-69:drake-build dongdong$ python -c 'import pydrake; print(pydrake.__file__)'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/opt/drake/lib/python2.7/site-packages/pydrake/__init__.py", line 32, in <module>
    from . import common
  File "/opt/drake/lib/python2.7/site-packages/pydrake/common/__init__.py", line 3, in <module>
    from ._module_py import *
ImportError: dlopen(/opt/drake/lib/python2.7/site-packages/pydrake/common/_module_py.so, 2): Symbol not found: __PyThreadState_Current
  Referenced from: /opt/drake/lib/python2.7/site-packages/pydrake/common/_module_py.so
  Expected in: flat namespace
 in /opt/drake/lib/python2.7/site-packages/pydrake/common/_module_py.so

I followed the installation procedure with Mac. Can you help me out?

EricCousineau-TRI, 2019-03-25:
Can you print the results of `which python` to ensure that you're using the Homebrew version?
https://drake.mit.edu/python_bindings.html#check-installation

> [image]

EricCousineau-TRI, 2019-03-25:
Also, as a BTW, there are instructions to install via Docker if you're looking to just peruse the examples / do the assignments:
http://underactuated.csail.mit.edu/Spring2019/install_drake_docker.html

> [image]

EricCousineau-TRI, 2019-04-04:
Howdy @ddliugit! Can I ask if you've had a chance to look at this?

54WC, 2019-05-01:
I've got the same issue. Can you help me out? 

MacBook-Pro:~ wc$ which python
/usr/local/opt/python/libexec/bin/python  

MacBook-Pro:~ wc$ python --version
Python 3.7.3

jamiesnape, 2019-05-01:
@54WC Remove `/usr/local/opt/python/libexec/bin/python` from your `PATH`.

54WC, 2019-05-02:
@jamiesnape , thank you, it works for python -c 'import pydrake; print(pydrake.__file__)'. 
But running example under underactuated/src suffer issue of "ImportError: No module named _multiarray_umath" 

I find a similar error report from : https://github.com/numpy/numpy/issues/11254
the problem was solved: It was due to the fact that the pickle file was written on a Windows machine, and I was reading it on a Mac. Writing the file on the Windows in binary format (pickle 'wb' instead of 'w') solved this problem. 

Could you advice how to fix it under this situation?  Thanks! 
 

MacBook-Pro:~ wc$ echo $PYTHONPATH
/opt/drake/lib/python2.7/site-packages:/Users/wc/underactuated/src:

MacBook-Pro:~ wc$ echo $PATH
/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin

MacBook-Pro:~ wc$ python -c 'import pydrake; print(pydrake.__file__)'
/opt/drake/lib/python2.7/site-packages/pydrake/__init__.pyc

Weide-MacBook-Pro:src wc$ python double_pendulum/simulate.py
ImportError: No module named _multiarray_umath
Traceback (most recent call last):
  File "double_pendulum/simulate.py", line 6, in <module>
    from underactuated import FindResource, PlanarSceneGraphVisualizer
  File "/Users/wc/underactuated/src/underactuated/__init__.py", line 2, in <module>
    from planar_rigid_body_visualizer import PlanarRigidBodyVisualizer
  File "/Users/wc/underactuated/src/underactuated/planar_rigid_body_visualizer.py", line 12, in <module>
    import scipy.spatial
  File "/usr/local/lib/python2.7/site-packages/scipy/spatial/__init__.py", line 98, in <module>
    from .kdtree import *
  File "/usr/local/lib/python2.7/site-packages/scipy/spatial/kdtree.py", line 8, in <module>
    import scipy.sparse
  File "/usr/local/lib/python2.7/site-packages/scipy/sparse/__init__.py", line 231, in <module>
    from .csr import *
  File "/usr/local/lib/python2.7/site-packages/scipy/sparse/csr.py", line 15, in <module>
    from ._sparsetools import csr_tocsc, csr_tobsr, csr_count_blocks, \
ImportError: numpy.core.multiarray failed to import

54WC, 2019-05-02:
After remove numpy, and reinstall drake. "ImportError: No module named _multiarray_umath" is fixed. 
 double_pendulum/simulate.py works now.

EricCousineau-TRI, 2019-05-02:
Glad that worked!

@jamiesnape Can I ask why removing that path fixed the issue?
Does this mean we should update / clarify our instructions?

jamiesnape, 2019-05-02:
> Can I ask why removing that path fixed the issue?

Because it added a Python 3 executable named `python` contrary to PEP 394.

> Does this mean we should update / clarify our instructions?

No, this is not a usual situation. We can't cover unusual corner cases.

closed 2019-05-02 by jamiesnape