RobotLocomotion/drake #11007: `the Exponent -1 should be a positive integer but it is not the case.`

RussTedrake, 2019-03-25:
following up on the conversation from drake developers:
https://drakedevelopers.slack.com/archives/C0JNB2E3S/p1551753164015700

here is the first reproduction (from @TobiaMarcucci):
```
import numpy as np
from pydrake.all import MathematicalProgram
prog = MathematicalProgram()
A = np.ones((2,2))
x = prog.NewIndeterminates(2, 'x')
V = x.dot(A).dot(x)
print V.Jacobian(x)
```

here is another one, from me
```
from pydrake.all import MathematicalProgram, Jacobian, Evaluate
prog = MathematicalProgram()
x = prog.NewIndeterminates(2,'x')
f = np.array([-x[1], x[0] + x[1]*(x[0]*x[0]-1)])
V = x.dot(x)
Vdot = V.Jacobian(x).dot(f)
print Evaluate(Jacobian(Vdot.Jacobian(x), x), dict(zip(x,[0,0])))
```

we currently believe that calling `Expand` on the `symbolic::Expression` can resolve the issue, but the current API would require us to loop through each element of the expression.  This is very cumbersome, especially as it is not happening in code that i'm writing which should support arbitrary locally stable polynomial systems.

Opening this to track it.  I understand that it might not be very simple/elegant to fix.

soonho-tri, 2019-03-26:
1) Using `Vdot = V.Jacobian(x).dot(f).Expand()` instead of `Vdot = V.Jacobian(x).dot(f)` gives us the right answer:

```python
from pydrake.all import MathematicalProgram, Jacobian, Evaluate
prog = MathematicalProgram()
x = prog.NewIndeterminates(2,'x')
f = np.array([-x[1], x[0] + x[1]*(x[0]*x[0]-1)])
V = x.dot(x)
Vdot = V.Jacobian(x).dot(f).Expand()   # Previously, Vdot = V.Jacobian(x).dot(f)
print Evaluate(Jacobian(Vdot.Jacobian(x), x), dict(zip(x,[0,0])))
```

```
[[ 0.  0.]
 [ 0. -4.]]
```

I'll check the `Differentiate` function to use `/ X` instead of `* pow(X, -1)`. I think it should help for common cases.

RussTedrake, 2019-03-30:
n.b. -- i'm blocked again by this.  hitting it in basically everything that I try.

PR #11099 (2019-03-31): Use distributive law to handle differentiation of products
Close #11007

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11099)
closed 2019-03-31 by RussTedrake