RobotLocomotion/drake #11009: models: De-duplicate and clean up

EricCousineau-TRI, 2019-03-25:
Relates #10531

We have some redundant models. I want to remove redundancy, and make things clearer (and possibly rely on upstream models when possible).

For example, @robert-verkuil ran into an issue where there are two Acrobot model files in the Drake source tree *that he can't use* from CMake (in `underactuated`):
`FindResourceOrThrow("drake/multibody/benchmarks/acrobot/acrobot.sdf") ` indicates that this file is not installed
`FindResourceOrThrow("drake/examples/acrobot/Acrobot.sdf")` has an invalid spatial inertia for MBP's.

(Acute solution for this is to install `benchmarks/.../acrobot.sdf` for now.)

jwnimmer-tri, 2019-03-25:
See also #5408 about the acrobot model file vs its URDF counterpart.
See also #10294 for same disease in acrobot, but about the code not the model files.

EricCousineau-TRI, 2019-03-25:
Additionally: We have `//manipulation/models` and `//multibody/models`. One of these should disappear.

(Also, potentially we just move any generic models into `//models`? ...)

jwnimmer-tri, 2019-03-25:
> Additionally: We have //manipulation/models and //multibody/models. One of these should disappear.

On principle, I think it's _completely fine_ to have a distinct models folder in each of those packages.  The two packages have different purposes and audiences, so could plausibly have different model sets.

However, in particular if you want to argue that box and box_small are stupid things to have in multibody/models, then that's a-ok by me to fix that acute problem.

> (Also, potentially we just move any generic models into //models? ...)

This seems worse on all counts.

EricCousineau-TRI, 2020-04-10:
FYI @IanTheEngineer

jwnimmer-tri, 2020-04-11:
I'm not clear on what's still to be done here.  We already have more focused issues filed about Acrobot.  Do we have other examples of duplication or required clean-ups?  Or is the issue about seeking out any other similar problems, but we don't know of any yet?

jwnimmer-tri, 2020-09-29:
Ping @EricCousineau-TRI can we close this now?  Or else can you clarify what's needed?

EricCousineau-TRI, 2020-09-29:
Closing. #10531 is the other more general thing that needs fixing, but that's already tracked. The acrobot-specific bits are also tracked.

Thanks for the ping!

closed 2020-09-29 by EricCousineau-TRI