RobotLocomotion/drake #11010: install_data: Make explicit, remove unused cruft

EricCousineau-TRI, 2019-03-25:
(Warning: Rant-y)

At present, `install_data` implicitly names its target `models`. I've always been confused by this, and I want that to change.

Additionally, there are *zero* usages of any customization of `prod_models_prefix` or `test_models_prefix`. Those should be explicitly defined too, rather than arbitrary prefixes. The `extra_prod_models` story also makes me sad.

This should be broken into two parts:
* Model globbing should be consolidated, to permit globbing install + test data. (Test data isn't even installed; why is it even part of `install_data`???)
* Installation for models should be consolidated.

jwnimmer-tri, 2019-04-18:
> At present, install_data implicitly names its target models.

FYI In an intermediate revision of #11089, the author mistakenly added a new redundant filegroup to examples/pendulum, presumably because the existing install_data filegroup was so obscured as to be unknown.

So +1 on resolving this pain point.

jwnimmer-tri, 2023-01-27:
For clarity, here's the kind of patch we're looking for:

```patch
--- a/manipulation/models/iiwa_description/BUILD.bazel
+++ b/manipulation/models/iiwa_description/BUILD.bazel
@@ -4,7 +4,7 @@ load(
     "@drake//tools/skylark:drake_cc.bzl",
     "drake_cc_googletest",
 )
-load("//tools/install:install_data.bzl", "install_data")
+load("//tools/install:install_data.bzl", "models_filegroup", "install_data")
 load("//tools/lint:lint.bzl", "add_lint_tests")
 
 # This package is public so that other packages can refer to
@@ -13,6 +13,15 @@ package(
     default_visibility = ["//visibility:public"],
 )
 
+models_filegroup(
+    name = "models",
+    # testonly_filegroup_name = "test_models",  # N.B. Not needed here, no test_models.
+    extra = [
+        "LICENSE.TXT",
+        "iiwa_stack.LICENSE.txt",
+    ],
+)
+
 # === test/ ===
 
 drake_cc_googletest(
@@ -27,10 +36,7 @@ drake_cc_googletest(
 )
 
 install_data(
-    extra_prod_models = [
-        "LICENSE.TXT",
-        "iiwa_stack.LICENSE.txt",
-    ],
+    data = [":models"],
 )
 
 add_lint_tests()
```

The goal is that all target names are literals in the BUILD file, i.e., `name = "models"`, and that silly intermediate labels like `prod_models` go away.

jwnimmer-tri, 2023-03-01:
This is getting in my way for #15774.  I'll do it tomorrow.

PR #18923 (2023-03-02): Tease apart models_filegroup vs install_data
Delete unused `PendulumWithFriction.urdf`.

---

I've tested locally and the list of installed files is the same between this PR and the master branch.

Closes #11010.

Towards #15774.

FYI https://drake.mit.edu/stable.html#model-files

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/18923)
closed 2023-03-02 by EricCousineau-TRI