RobotLocomotion/drake #11012: parsing: Support model deprecation

EricCousineau-TRI, 2019-03-25:
Motivated by #11009: It'd be nice to actively warn users if a model is deprecated, because I want to delete confusingly redundant files.

For now, we can a text-file scanning heuristic that will scan for a `DEPRECATED` comment and provide the message. Later, we can consider adding a `<drake:deprecated/>` tag for SDF (and maybe URDF).

jwnimmer-tri, 2019-03-25:
I'd say make it work for any resource, not just models.

We could install them to `share/drake/deprecated` instead (or whatever) once they become deprecated, and FindResourceOrThrow can be taught to look there and warn (but succeed).  No need to pollute the parsers or otherwise with an installation-related concern.

EricCousineau-TRI, 2019-03-25:
SGTM. Updated title; would be nice to ensure it works in concert with `package://` discovery (#10531), at least for Drake's package path file resolution.

EricCousineau-TRI, 2019-03-25:
Hm... If we do it purely in a resource-based fashion, how might we add a message for deprecation timeline and what to use instead? I guess we could do something like:
```
${install}/share/drake/.deprecated/
   examples/acrobot/
     Acrobot.sdf
     .Acrobot.sdf.deprecation_message.txt
```

That way it spews a message on access to that resource?

jwnimmer-tri, 2019-03-25:
Sure, or even a single YAML/JSON file or whatever, where we can list files + messages (sharing a message with multiple files is probably common).  Doing that might even mean that we can keep the files in their original path, but still have FindResource warn?  Less shuffling is probably better.

jwnimmer-tri, 2020-05-14:
@EricCousineau-TRI Now that we have monthly release notes, I wonder if mentioning pending resource removals there is a sufficient solution?

EricCousineau-TRI, 2020-05-14:
Yeah, I'd be fine with that!

jwnimmer-tri, 2020-05-14:
=> #10605 for process note, but nothing left to do here.

closed 2020-05-14 by jwnimmer-tri