RobotLocomotion/drake #11013: MultibodyPlant<double> computes Jdot * v

hongkai-dai, 2019-03-25:
The term Jdot * v is often needed when computing the time derivative of J * v. Currently to compute Jdot * v, we have to use MultibodyPlant<ADS> to compute `J`, with its gradient `dJdq` as the gradient of `J`, and then compute dJdq * qdot to get Jdot, and then compute Jdot * v. This computation is complicated and not efficient. It is better to compute Jdot * v directly from MBP<double>.

This issue was originally reported in #466. We had the function to compute Jdot * v in RigidBodyPlant, it would be good to add that feature back to MBP.

amcastro-tri, 2019-03-25:
@hongkai-dai, is [this](https://github.com/RobotLocomotion/drake/blob/master/multibody/plant/multibody_plant.h#L1742)? or maybe another variant? 
In MBP we call "bias" to the product `Jdot * v`, maybe that caused confusion? see #10664.

hongkai-dai, 2019-03-25:
Ah, I didn't realize it is called "bias". Yes, that is what we need. I will close this issue.

closed 2019-03-25 by hongkai-dai