RobotLocomotion/drake #11014: examples: Merge `//examples/multibody/*` into `//examples`

EricCousineau-TRI, 2019-03-25:
At present, a *lot* of examples use `RigidBodyTree`. We should delete them or move them to `attic` (I prefer deletion!).

We should then merge `//examples/multibody/*` into `//examples` (as it's no longer a useful distinction?).

\cc @jwnimmer-tri @amcastro-tri 

Relates #5408

jwnimmer-tri, 2020-12-09:
> At present, a lot of examples use RigidBodyTree. We should delete them or move them to attic (I prefer deletion!).

All of the RBT stuff is gone now (either ported, or deleted).

\CC @rpoyner-tri @joemasterjohn FYI we should not be adding _new_ subfolders to `//examples/multibody/...`.

rpoyner-tri, 2020-12-09:
The only folder names that conflict between //examples and //examples/multibody are acrobot and pendulum. If those cases could be resolved somehow, we could just hoist everything and delete the directory.

Does such an operation require some sort of deprecation cycle?

jwnimmer-tri, 2020-12-09:
Almost nothing is available for public consumption:

```
-*- mode: grep; default-directory: "~/jwnimmer-tri/drake/examples/multibody/" -*-
Grep started at Wed Dec  9 15:34:34

find . -type f -exec grep --color -nH -e //visibility {} +
./cart_pole/BUILD.bazel:17:package(default_visibility = ["//visibility:public"])
./rolling_sphere/BUILD.bazel:4:package(default_visibility = ["//visibility:public"])
./cassie_benchmark/BUILD.bazel:11:package(default_visibility = ["//visibility:public"])

find . -type f -exec grep --color -nH -e install {} +
./cart_pole/BUILD.bazel:10:load("//tools/install:install_data.bzl", "install_data")
./cart_pole/BUILD.bazel:58:install_data()
```

I don't immediately see anything that would be risky to rename without deprecation, but it's worth a second look.

amcastro-tri, 2020-12-10:
I personally like having an `/examples/multibody` directory since that way all the MBP related examples can be found in a single place. Is it only me? otherwise I find a monolithic `examples/` directory to be a soup without much of an ordering.

jwnimmer-tri, 2020-12-10:
Maybe it helps you, but the extra level of nesting confuses everyone else.  We can add an index table to the documentation to help categorize the examples (across several axes), but the current unprincipled folder split is not a good way to do it.

amcastro-tri, 2020-12-10:
Definitely it helps me, surprised that it doesn't help anyone else? probably a good survey for Drake devs?
The option of some soft of index would be great also. Would that be in the form of a README file?

jwnimmer-tri, 2020-12-10:
Yes, we'd probably benefit from either a README.md, or more likely an `examples_doxygen.h`, that highlighted what was available, what features of Drake it tickled, what programming languages it used, maybe some videos, etc.