RobotLocomotion/drake #11027: Use --build_tests_only option in slow pre-merge CI builds

jwnimmer-tri, 2019-03-26:
There is a flag https://docs.bazel.build/versions/master/command-line-reference.html#flag--build_tests_only where only tested code gets compiled.  This is likely to speed up the pre-merge CI builds for large-footprint builds with symbols, such as lsan.  We should see if turning this on makes things better, and if we agree that it's okay to lose the "prove this untested code can compile in lsan mode" is any great loss.

Probably turning it on for all of the foosan makes sense (in both pre- and post-merge).  We should probably _not_ turn it on for the vanilla release and debug builds, since we want those to always compile, even for untested code.

jwnimmer-tri, 2019-03-26:
(Might also make sense for coverage, valgrind, etc.)

EricCousineau-TRI, 2019-04-04:
Talking to Jamie, he can try this out briefly on `drake-ci`.
Would you be up for this being assigned to Jamie?

jamiesnape, 2019-04-04:
https://github.com/RobotLocomotion/drake-ci/commit/7dfa5869a49c2b15028cc8256a888f56c11e5dfb

jwnimmer-tri, 2019-04-04:
Shouldn't this be a `dynamic_analysis/bazel.rc` flag, not CI magic?  I don't think users should have to compile the useless tests, either?

jwnimmer-tri, 2019-04-04:
(Or is this just a timing test to see if it's worth it, at which point we roll back into Drake?)

jamiesnape, 2019-04-04:
> (Or is this just a timing test to see if it's worth it, at which point we roll back into Drake?)

Indeed.

jamiesnape, 2019-04-04:
We just had the quickest ASan job (out of the last 300 jobs) by 4 minutes, so looks promising.

jamiesnape, 2019-04-04:
I will check the data again tomorrow morning (in particular I want to check the nightlies), but looks like it is going to be worth enabling for developers too, so I will PR that tomorrow and revert the CI change.

PR #11178 (2019-04-10): Add --build_test_only to entries in dynamic analysis bazel.rc
And other cleanup. The choice of `build:` and `test:` prefixes seems arbitrary; all those options in this file are valid for both commands and inherited by `test` when set as `build`. Also `msan` and `msan_everything` have never  been functional, and it seems unlikely that they even will be. https://github.com/RobotLocomotion/drake-ci/commit/7dfa5869a49c2b15028cc8256a888f56c11e5dfb should be reverted once this merges.

Closes #11027.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11178)
closed 2019-04-10 by jamiesnape