RobotLocomotion/drake #11035: Allow post-declaration modification of cache entry prerequisites

SeanCurtis-TRI, 2019-03-26:
While some systems' components are compile-time defined, some systems are defined dynamically (changing number of ports, state, etc.). If these dynamic systems have output ports, cache entries, whatever, that depend on those dynamically-created structures, there is not currently many good options for doing so, including 1) introduce a "finalize" invocation, 2) declaring broad pre-requisites that may encompass more than is necessary to guarantee capturing the necessary.

If, instead, we had the ability to grab a *declared* object with pre-requisites and add newly added tickets to its prerequisite set, we wouldn't need finalize or big-hammer pre-requisite declarations.

jwnimmer-tri, 2019-03-26:
For any solution here, I'd like to encourage authors to have meaningful names and control flow.  So for example, declare a ticket "All input ports that give us poses", have the downstream computations depend on that ticket (which can happen early), and then later as we add new input ports we should have them feed into the "Input ports with poses" ticket.  We should **not** have the "add input port" method go and add its ticket to a bunch of sundry calc methods / cache entries.

SeanCurtis-TRI, 2019-03-26:
I love that. It maintains maximum clarity at the point of declaration.

sherm1, 2019-03-26:
I believe this is already possible! The outline is:
- in System constructor, create a new DependencyTracker and get the ticket
- give the ticket a friendly name as Jeremy suggests above
- declare the cache entry and make it depend on that ticket
- as you create new input ports, subscribe the new tracker to those ports' tickets

I think all the necessary methods are public. We'll try this and upgrade if necessary.

sherm1, 2019-03-26:
I spoke to soon. Only the dependency tickets are present in the System -- the trackers exist only in the Context so can't be created in a System constructor. It wouldn't be too hard to support custom dependency trackers, but it would be a project.

What would be possible now is to use the cache entry itself this way (it actually _does_ hold a "custom dependency tracker" of sorts). Give it a friendly name indicating what it will depend on, just as you would have done for a custom tracker. Then add the dependencies (actually called "prerequisites") to it later.

PR #11049 (2019-03-28): Permit late addition of prerequisites to already-declared cache entries
Normally all cache entry prerequisites are provided at the time a cache entry is declared. That is difficult in
some Systems with complicated construction sequences, such as SceneGraph (see #11035).

This PR adds an `(Advanced)` method to CacheEntry that permits modification of the prerequisites set after initial declaration. This is not technically difficult because a CacheEntry simply maintains an `std::set` of ticket numbers that is not used until Context creation time. Modifying that set any time prior to that is fine. [... cut]
closed 2019-03-28 by soonho-tri