RobotLocomotion/drake #11037: Build errors from Apple LLVM version 10.0.1 (clang-1001.0.46.3)

soonho-tri, 2019-03-26:
Some examples:

```
In file included from bindings/pydrake/systems/lcm_py_bind_cpp_serializers.cc:6:
bazel-out/darwin-dbg/bin/bindings/pydrake/systems/_virtual_includes/lcm_pybind/drake/bindings/pydrake/systems/lcm_pybind.h:38:10: error: local variable 'py_cls' will be copied despite being returned by name [-Werror,-Wreturn-std-move]
  return py_cls;
         ^~~~~~
bindings/pydrake/systems/lcm_py_bind_cpp_serializers.cc:17:3: note: in instantiation of function template specialization 'drake::pydrake::pysystems::pylcm::BindCppSerializer<robotlocomotion::image_array_t>' requested here
  BindCppSerializer<robotlocomotion::image_array_t>("robotlocomotion");
  ^
```

```
external/local_config_cc/wrapped_clang '-D_FORTIFY_SOURCE=1' -fstack-protector -fcolor-diagnostics -Wall -Wthread-safety -Wself-assign -fno-omit-frame-pointer -g '-std=c++11' -iquote . -iquote ... (remaining 142 argument(s) skipped)

Use --sandbox_debug to see verbose messages from the sandbox
common/test/reset_on_copy_test.cc:159:5: error: explicitly assigning value of variable of type 'reset_on_copy<int>' to itself [-Werror,-Wself-assign-overloaded]
  w = w;
  ~ ^ ~
```

soonho-tri, 2019-03-26:
I have https://github.com/soonho-tri/drake/commit/58f8ca5c8c45869ff1899f01bab85d92eed668d7 which fixes the problem for the new Apple clang. But it generates warnings when I tried it with an old version of clang (i.e. `clang-6.0`):

```
INFO: From Compiling common/test/symbolic_polynomial_matrix_test.cc:
warning: unknown warning option '-Wno-self-assign-overloaded'; did you mean '-Wno-self-assign-field'? [-Wunknown-warning-option]
warning: unknown warning option '-Wno-return-std-move' [-Wunknown-warning-option]
2 warnings generated.
```

jamiesnape, 2019-03-27:
We can add Apple specific flags. The `COMPILER_ID` use can use in https://github.com/RobotLocomotion/drake/blob/f3a152c11b8a4f607093bc83009262cca373d392/tools/skylark/drake_cc.bzl#L56-L62 should be `AppleClang`.

jamiesnape, 2019-03-27:
Something like
```python
APPLECLANG_FLAGS = ["-Wno-return-std-move", "-Wno-self-assign-overloaded"] 

 if COMPILER_ID.endswith("Clang"): 
     result = CLANG_FLAGS + rule_copts + rule_clang_copts
     if COMPILER_ID == "AppleClang":
          result = result + APPLECLANG_FLAGS
 elif COMPILER_ID == "GNU": 
     extra_gcc_flags = GCC_CC_TEST_FLAGS if cc_test else [] 
     result = GCC_FLAGS + extra_gcc_flags + rule_copts + rule_gcc_copts 
 else: 
     result = rule_copts
```

jwnimmer-tri, 2019-03-27:
It's probably worth adding that structure into the file, even if the APPLECLANG_FLAGS soon goes to empty after we tidy up Drake's code nits.

jamiesnape, 2019-03-27:
Agreed.

soonho-tri, 2019-03-27:
I've updated my WIP branch: https://github.com/soonho-tri/drake/commits/fix-11037

 - Updated the code so that we don't need to suppress `return-std-move`.
 - Fixed a couple of locations not to generate `self-assign-overloaded`
 - The only place where we get `self-assign-overloaded` is in `symbolic_py.cc` which uses pybind11 idiom `def(py::self /= py::self)`.

BTW, `self-assign-overloaded` is also added to clang-8 so this is not really apple-specific.

jamiesnape, 2019-03-27:
I think we can also do version-specific flags, but that is a load of extra code that we should probably defer or workaround.

jamiesnape, 2019-03-27:
I would say PR https://github.com/soonho-tri/drake/commit/809598265f613ae7aa26d788037a03844af0ed2a and the addition `-Wno-self-assign-overloaded` for Mac only for now, but others may disagree.

soonho-tri, 2019-03-27:
Yes, I'm fixing it up..

PR #11060 (2019-03-28): Handle return-std-move and self-assign-overload warnings from Apple LLVM 10.0.1
Close #11037

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11060)
closed 2019-03-28 by jamiesnape