RobotLocomotion/drake #11051: suspicious behavior in python CalcGravityGeneralizedForces

RussTedrake, 2019-03-27:
https://stackoverflow.com/questions/55373650/how-to-properly-use-multibodyplantcalcgravitygeneralizedforces-in-pydrake

sherm1, 2019-03-27:
One thought that pops to mind @amcastro-tri is that the PyDrake result is consistent with gravity=0. That makes me wonder whether the fancy numpy binding for UniformGravity might be ignoring the given vector.

gizatt, 2019-03-27:
Maybe this is the same as what you mean, but it looks like MBT has two methods for registering ForceElements -- one for generic force elements, and one for UniformGravityFieldElements. Maybe the Python call is not going into the specialized (latter) method and not getting registered to the MBT's `gravity_field_`, which is what gets checked in the `CalcGravityGeneralizedForces` method?

amcastro-tri, 2019-03-28:
yes, that's what's happening. Let me see how to fix.

amcastro-tri, 2019-03-28:
In C++ we also have `UniformGravityFieldElement::CalcGravityGeneralizedForces()`. We could add the python binding for that so you can call that computation directly on the element. Would that work @gizatt?

jwnimmer-tri, 2019-03-28:
Adding Gravity should work the same in both C++ and broken.  To the extent C++ uses a specialization to achieve this, pydrake must deal with that as well.  Even if `UniformGravityFieldElement::CalcGravityGeneralizedForces` also gets bound, `AddForceElement` is still broken.

EricCousineau-TRI, 2019-03-28:
Solving the API parity can be done by adding a overlay (before the first `AddForceElement`) that catches `UniformGravityFieldElement` and dispatches to that specialization.

I can submit this PR if you'd like.

TBH, though, it'd be *way* nicer if the public API didn't require internal compile-time specialization...
The better solution is to not constrain `AddForceElement` to only permit one addition of `UniformGravityForceElement`, and instead it should scan the list after-the-fact to get the gravity... Or perhaps remove `CalcGravityGeneralizedForces` and instead call it `CalcConservativeGeneralizedForces`?

Also, the usages of `gravity_field_` seem really weird. It's used for `CalcGravityGeneralizedForces` (find I guess...) and as a heuristic for `set_penetration_allowance`... with a default value???

amcastro-tri, 2019-03-28:
> as a heuristic for set_penetration_allowance... with a default value???

I agree, that'll get resolved with future better contact models.

amcastro-tri, 2019-03-28:
@EricCousineau-TRI gave my a lil snippet of the solution for me to try. I'll try to PR soon. Thanks!

amcastro-tri, 2019-03-28:
Fix in PR #11069.

PR #11069 (2019-03-30): Implements binding to add a gravity field
Resolves #11051.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11069)
closed 2019-03-30 by RussTedrake