RobotLocomotion/drake #11059: Discrete updates in Simulator invalidates all subsystems in a Diagram at the rate of the fastest updating discrete system

edrumwri, 2019-03-27:
The Simulator performs a discrete update on a Diagram at the fastest update rate of any subsystem. *All* discrete values in *all* subsystems get updated at this rate, which means that all computations cached on the discrete variables become invalidated (even if they don't really change). This is liable to be a significant performance problem. Marking the priority as "medium" but I could see it quickly becoming "high".

Related to #10149 (copy-on-write would address this problem in both issues).

jwnimmer-tri, 2019-03-27:
I remain 100% convinced that we should solve this (and #10149) by making Diagram do smarter things (not Cloning too much, not Invalidating too much) when only some subsystems have update events happening.  It really, really, isn't that hard.  Can I convince you to try that first, before unsafe raw events and before CoW affordances bleed everywhere?

sherm1, 2019-03-27:
I'm happy to give that a try before we go whole-COW!

jwnimmer-tri, 2019-03-27:
(I am happy to do a quick VTC when relevant to explain my reasoning, if that helps.)

sherm1, 2019-03-28:
One CoW-free approach that could partially address the problem: 
- teach Diagram dispatchers to copy only the _structure_ of Diagram DiscreteValues (DVs) and State until reaching LeafSystem. Then copy DVs or States only for LeafSystems that have pending Events, just prior to dispatching those. The result would be partially-filled-in DiagramDiscreteValues or DiagramState objects.
- provide context.SetDiscreteState(updated_dvs) and SetState(updated_state) methods for the Simulator to use to perform the updates. These would understand how to interpret the missing pieces as "don't update this part".

This has the disadvantage that we still have to copy all DVs or the entire State for any LeafSystem with a pending Event, even if the event handler ends up just touching one state variable, or perhaps even taking no action. For full backwards compatibility we have to do that because we promise to provide allocated objects for the event handlers to write into. With CoW those objects are "borrowed" and copied only if written to; here they are copied unconditionally.

Still if the above could work it would decouple separate subsystems from one another so that events happening in one subsystem wouldn't cause copying and invalidations in other subsystems. I think that might avoid some of the worst cases.

jwnimmer-tri, 2019-03-28:
> ... it would decouple separate subsystems from one another so that events happening in one subsystem wouldn't cause copying and invalidations in other subsystems. I think that might avoid some of the worst cases.

I think you're right that partitioning by System is good enough for now.  I am not worried about a System that has multiple groups of state and we're copying all of it when we only needed to copy some of it -- our problems occur either when high-rate small-state systems cause low-rate large-state systems to copy unnecessarily, or when we get the O(N^2) problem of N systems each with an update so we have N^2 copies.  And anyway, once we have the per-System problem solved, the substate problem is easily solved as well through event-declaration sugar, where the Declare callback on the update is typed to give you your unerased `Value<T>* next_value` (or writeable Eigen block, etc.) instead of your whole `State* next_state`.

> One CoW-free approach that could partially address the problem ...

I think the Diagram fixup is even easier than you are imagining with skeleton state and whatever.  It's really simple: when we have a batch of updates to compute at the same time, we allocate (or reuse) storage for each individual update's output argument.  Then we call all of their CalcUpdate methods to populate the new values.  Then we copy the new values into the current (mutable) Context's state.  It's just a simple transaction -- compute first and _then_ apply, so that compute isn't changed by intermediate applications.  To the extent the intra-framework APIs defeat this methodology, they are bad and we'll change them.  There is no reason to disturb the public APIs that users are interacting with -- they are sufficient to formulate the required computations, since nothing in the LeafSystem update-declaration API needs to know about how diagrams work; we already only give it a pointer to a mutable sub-State.

Still open to VTC if this isn't clear enough.

jwnimmer-tri, 2019-03-28:
Hmm, is this issue a duplicate of #10878?

amcastro-tri, 2019-03-28:
To bring what was lost in #10878 (and possibly Slack conversations). This **is** the cause for the current slow donw from 1.6X to 1.15X of an Anzu sim (most likely similar slow downs in similar ones).

amcastro-tri, 2019-03-28:
My 2 cents given I really have no experience with the COW approach:
I think I agree with @jwnimmer-tri, at least for a shorter term solution. Most of my readings about COW pointers online show it as a curiosity with not really many (any?) success stories. Mostly, I only found warnings from people who tried it saying it over-complicates things with way too many levels of indirection (we have lots of those already!). 

It might also be that is way easier to implement than what I think? if so, I propose to do a quick (fancyness asides please! no docs, not sugar, no thread safe) prototype within the next couple days and see if there are any bottlenecks in either the COW or the by-system-partitioning approach?

sherm1, 2019-03-28:
Notes from f2f w/Jeremy regarding the CoW-free subsystem decoupling described above: 
- copying State tree structure only provides a place to hang updated leaf States and easy traversal, but could be expensive
- would be nice just to collect the updated leaf States and then apply them
- one possibility is to add something like `EventCollection::Update(&diagram)` after the States are collected; the event collection already has exactly the right structure and is populated only where there are actual events
- a remaining difficulty is how/where to hold on to the leaf State update objects as they are accumulated

PR #11113 (2019-04-02): Add System methods for selective state updates after an event
Provides `System::ApplyDiscreteVariableUpdate()` and `ApplyUnrestrictedUpdate()` methods guided by the Event collection that was used by the Simulator in `CalcDiscreteVariableUpdates()` or `CalcUnrestrictedUpdate()` to do a minimal amount of state updating.

Resolves #11059
Relates #10149

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11113)
closed 2019-04-02 by sherm1